tzapu/WiFiManager #1580: IP,GetWay and SubNetMask set to :0.0.0.0

Mehdi-Dehghan, 2023-04-02:
## PLEASE TRY Latest Master BRANCH before submitting bugs, in case they were already fixed. ##

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32: ESP32

Hardware:		ESP32 240MHz, 520KB RAM, 4MB Flash

Core Version:      	2.0.15-rc.1

### Description

Problem description

### Settings in IDE

Module:  			ESP32-DevKitC

Additional libraries: 

### Sketch

```cpp
#BEGIN
#include <Arduino.h>

void setup() {
  wifiManager.setAPCallback(configModeCallback);
  wifiManager.setWebServerCallback(bindServerCallback);
  wifiManager.setSaveConfigCallback(saveWifiCallback);
  wifiManager.setSaveParamsCallback(saveParamCallback);
  wifiManager.setShowStaticFields(true); // force show static ip fields

  //start-block1
  IPAddress _ip,_gw,_sn;
  _ip.fromString(static_ip);
  _gw.fromString(static_gw);
  _sn.fromString(static_sn);
  //end-block1
  wifiManager.setSTAStaticIPConfig(_ip,_gw,_sn);
  wifiManager.setConfigPortalTimeout(60);
  
  wifiManager.autoConnect("MyTest");
  
  Serial.println("---------First Step---------");
  Serial.println(WiFi.localIP());
  Serial.println(WiFi.gatewayIP());
  Serial.println(WiFi.subnetMask());
}

void loop() {

}
#END
```

### Debug Messages

STA IP set:0.0.0.0
STA GetWay set:0.0.0.0
STA SubNetMask set:0.0.0.0
messages here
Hi

I tried to set IP, GetWay, subnetmask and also SSID, Password in portal. but All IP's set to 0.0.0.0
How can I cope this issue?

Thanks in advance

tablatronix, 2023-04-02:
Show me your variables for ips

Mehdi-Dehghan, 2023-04-03:
Thanks for your reply.
this is my complete code:
```cpp
//#include <ESP8266WiFi.h>
#include <WiFi.h>
#include <WiFiClient.h>
//#include <ESP8266WebServer.h>
#include <WebServer.h>
#include <SPIFFS.h>
#include <FS.h> // FOR LittleFS
#include <ctype.h> // for isNumber check
#include <WiFiManager.h> 

//---------------------------
//default custom static IP
char static_ip[16];// = "192.168.1.56";
char static_gw[16];// = "192.168.1.1";
char static_sn[16];// = "255.255.255.0";
//---------------------------------------------------------------
WebServer server(80);
WiFiManager wifiManager;
//---------------------------------------------------------------
//flag for saving data
bool shouldSaveConfig = false;

//callback notifying us of the need to save config
void saveConfigCallback () 
{
  Serial.println("Should save config");
  shouldSaveConfig = true;
  Serial.println(WiFi.localIP().toString());
  Serial.println(WiFi.gatewayIP().toString());
  Serial.println(WiFi.subnetMask().toString());
}

void saveWifiCallback()
{
  Serial.println("[CALLBACK] saveCallback fired");
}

//gets called when WiFiManager enters configuration mode
void configModeCallback (WiFiManager *myWiFiManager) 
{
  Serial.println("[CALLBACK] configModeCallback fired");
  // myWiFiManager->setAPStaticIPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0)); 
  // Serial.println(WiFi.softAPIP());
  //if you used auto generated SSID, print it
  // Serial.println(myWiFiManager->getConfigPortalSSID());
  // 
  // esp_wifi_set_bandwidth(WIFI_IF_AP, WIFI_BW_HT20);
}

void saveParamCallback()
{
  shouldSaveConfig = true;
  Serial.println("========================================");
  Serial.println("[CALLBACK] saveParamCallback fired");
  Serial.print("STA IP set:");
  Serial.println(WiFi.localIP());
  Serial.print("STA GetWay set:");
  Serial.println(WiFi.gatewayIP());
  Serial.print("STA SubNetMask set:");
  Serial.println(WiFi.subnetMask());
  wifiManager.setSTAStaticIPConfig(WiFi.localIP(), WiFi.gatewayIP(), WiFi.subnetMask());
  //save the custom parameters to FS
 
  Serial.println("========================================");
  // wm.stopConfigPortal();
}

void bindServerCallback()
{
  //wifiManager.server->on("/custom",handleRoute); // this is now crashing esp32 for some reason
  // wm.server->on("/info",handleRoute); // you can override wm!
}
//---------------------------------------------------------------
//---------------------------------------------------------------
///////////////////////////////////////
// first funtion to run at power up //
//////////////////////////////////////

void setup(void)
{
  // You can open the Arduino IDE Serial Monitor window to see what the code is doing
  Serial.begin(115200);

  wifiManager.setClass("invert"); // dark theme
  wifiManager.setDebugOutput(true);
  wifiManager.setAPCallback(configModeCallback);
  wifiManager.setWebServerCallback(bindServerCallback);
  wifiManager.setSaveConfigCallback(saveWifiCallback);
  wifiManager.setSaveParamsCallback(saveParamCallback);
  wifiManager.setShowStaticFields(true); // force show static ip fields

  IPAddress _ip,_gw,_sn;
  _ip.fromString(static_ip);
  _gw.fromString(static_gw);
  _sn.fromString(static_sn);
  
  wifiManager.setSTAStaticIPConfig(_ip,_gw,_sn);
  wifiManager.setConfigPortalTimeout(60);
  // wifiManager.startConfigPortal("OnDemandAP");
  wifiManager.autoConnect("MyTest");
  // wifiManager.persistent(true);
  
 if (shouldSaveConfig) 
  {
    shouldSaveConfig = false;
    Serial.println("saving config");
    Serial.println(WiFi.localIP());
    Serial.println(WiFi.gatewayIP());
    Serial.println(WiFi.subnetMask());
    //end save
  }
  
  if (WiFi.status() == WL_CONNECTED)
  {
    Serial.print("\n\r \n\rWorking to connect");
    Serial.print("Connected to ");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
    Serial.println(WiFi.gatewayIP());
    Serial.println(WiFi.subnetMask());
  }
  delay(10);

}

////////////////////////////////////////////////////////////////////////////////

void loop(void)
{

}
```

tablatronix, 2023-04-03:
And if you set these? what happens?

```
char static_ip[16];// = "192.168.1.56";
char static_gw[16];// = "192.168.1.1";
char static_sn[16];// = "255.255.255.0";
```

Mehdi-Dehghan, 2023-04-03:
> And if you uncomment these? what happens? you know you have to save these somewhere right? ESP does not
> 
> ```
> char static_ip[16];// = "192.168.1.56";
> char static_gw[16];// = "192.168.1.1";
> char static_sn[16];// = "255.255.255.0";
> ```

I uncommented and in portal showed those IPs. but when I changed in portal (192.168.4.1 adress) to another things they were still fixed:

load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
*wm:AutoConnect 
*wm:STA IP set: 192.168.1.56
*wm:Connecting to SAVED AP: New
*wm:connectTimeout not set, ESP waitForConnectResult... 
*wm:AutoConnect: FAILED for  3127 ms
*wm:StartAP with SSID:  MyTest
*wm:AP IP address: 192.168.4.1
[CALLBACK] configModeCallback fired
*wm:Starting Web Portal
*wm:1 networks found
[ 64610][E][WebServer.cpp:649] _handleRequest(): request handler not found
[ 64636][E][WebServer.cpp:649] _handleRequest(): request handler not found
========================================
[CALLBACK] saveParamCallback fired
STA IP set:192.168.1.56
STA GetWay set:192.168.1.1
STA SubNetMask set:255.255.255.0
========================================
*wm:STA IP set: 192.168.1.56
*wm:Connecting to NEW AP: New
*wm:connectTimeout not set, ESP waitForConnectResult... 
*wm:[ERROR] Connect to new AP Failed 
*wm:config portal has timed out 
*wm:config portal exiting 
saving config
192.168.1.56
192.168.1.1
255.255.255.0

Mehdi-Dehghan, 2023-04-04:
I did several times (reconnect & save in flash). but that still has this issue.
I tried to change IPs!

```
//#include <ESP8266WiFi.h>
#include <WiFi.h>
#include <WiFiClient.h>
//#include <ESP8266WebServer.h>
#include <WebServer.h>
#include <SPIFFS.h>
#include <FS.h> // FOR LittleFS
#include <ctype.h> // for isNumber check
#include <WiFiManager.h> 
#include <ArduinoJson.h>
//---------------------------
//default custom static IP
char static_ip[16] = "192.168.1.7";
char static_gw[16] = "192.168.1.1";
char static_sn[16] = "255.255.255.0";
//---------------------------------------------------------------
WebServer server(80);
WiFiManager wifiManager;
//---------------------------------------------------------------
//flag for saving data
bool shouldSaveConfig = false;

//callback notifying us of the need to save config
void saveConfigCallback () 
{
  Serial.println("Should save config");
  shouldSaveConfig = true;
  Serial.println(WiFi.localIP().toString());
  Serial.println(WiFi.gatewayIP().toString());
  Serial.println(WiFi.subnetMask().toString());
}

void saveWifiCallback()
{
  Serial.println("[CALLBACK] saveCallback fired");
}

//gets called when WiFiManager enters configuration mode
void configModeCallback (WiFiManager *myWiFiManager) 
{
  Serial.println("[CALLBACK] configModeCallback fired");
  // myWiFiManager->setAPStaticIPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0)); 
  // Serial.println(WiFi.softAPIP());
  //if you used auto generated SSID, print it
  // Serial.println(myWiFiManager->getConfigPortalSSID());
  // 
  // esp_wifi_set_bandwidth(WIFI_IF_AP, WIFI_BW_HT20);
}

void saveParamCallback()
{
  shouldSaveConfig = true;
  Serial.println("========================================");
  Serial.println("[CALLBACK] saveParamCallback fired");
  Serial.print("STA IP set:");
  Serial.println(WiFi.localIP());
  Serial.print("STA GetWay set:");
  Serial.println(WiFi.gatewayIP());
  Serial.print("STA SubNetMask set:");
  Serial.println(WiFi.subnetMask());
  wifiManager.setSTAStaticIPConfig(WiFi.localIP(), WiFi.gatewayIP(), WiFi.subnetMask());
  //save the custom parameters to FS
 
  Serial.println("========================================");
  // wm.stopConfigPortal();
}
[... 138 lines cut ...]
*wm:connectTimeout not set, ESP waitForConnectResult... 
*wm:AutoConnect: FAILED for  60628 ms
*wm:StartAP with SSID:  MyTest
*wm:AP IP address: 192.168.4.1
[CALLBACK] configModeCallback fired
*wm:Starting Web Portal
*wm:1 networks found
========================================
[CALLBACK] saveParamCallback fired
STA IP set:192.168.1.7
STA GetWay set:192.168.1.1
STA SubNetMask set:255.255.255.0
========================================
*wm:STA IP set: 192.168.1.7
*wm:Connecting to NEW AP: New
*wm:connectTimeout not set, ESP waitForConnectResult...
*wm:[ERROR] Connect to new AP Failed 
*wm:config portal has timed out
*wm:config portal exiting 
192.168.1.7
192.168.1.1
255.255.255.0
saving config
{"static_ip":"192.168.1.7","static_gw":"192.168.1.1","static_sn":"255.255.255.0"}
```

Mehdi-Dehghan, 2023-04-04:
I like to say, befor using ESP32, I used ESP8266 and it hadn't any problem and worked well!

tablatronix, 2023-04-04:
Ok why is your connection failing though?

tablatronix, 2023-04-04:
There is something else going on here, 

`[ 38365][V][WebServer.cpp:296] handleClient(): New client: client.localIP()=0.0.0.0`

Let me figure out wtf is going on, looks like an esp bug

tablatronix, 2023-04-04:
oh I was stuck on tasmotas esp32 , had to force remove my platofrm in pio

tablatronix, 2023-04-05:
ugh

mounting FS...
E (675) SPIFFS: mount failed, -10025
[   685][E][SPIFFS.cpp:89] begin(): Mounting SPIFFS failed! Error: -1

tablatronix, 2023-04-05:
This is wrong.

```
  wifiManager.setSTAStaticIPConfig(WiFi.localIP(), WiFi.gatewayIP(), WiFi.subnetMask());
```
you should be passing in the parameters here

closed 2023-04-05 by tablatronix

Mehdi-Dehghan, 2023-04-05:
Thanks for your time.

tablatronix, 2023-04-05:
Get it working?

Mehdi-Dehghan, 2023-04-16:
Yes That works true.
Thanks a lot