tzapu/WiFiManager #1582: Wifi Disconnection during captive portal on XH-C3F sparkleiot

micdj, 2023-04-07:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32: 

Hardware:			XH-C3F

Core Version:      	2.0.7

### Description

During captive portal, in softAP with all basic examples, wifi disconnect from client on alla platform(windows, ios and android) but softAP is up, if I reconnect it start again and open captive portal automatically again, then could work or disconnect again.

### Settings in IDE

Module:  			
ESP32C3 Dev Module

### Sketch

#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

void setup() {
    // WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP
    // it is a good practice to make sure your code sets wifi mode how you want it.

    // put your setup code here, to run once:
    Serial.begin(115200);
    
    //WiFiManager, Local intialization. Once its business is done, there is no need to keep it around
    WiFiManager wm;

    // reset settings - wipe stored credentials for testing
    // these are stored by the esp library
    // wm.resetSettings();

    // Automatically connect using saved credentials,
    // if connection fails, it starts an access point with the specified name ( "AutoConnectAP"),
    // if empty will auto generate SSID, if password is blank it will be anonymous AP (wm.autoConnect())
    // then goes into a blocking loop awaiting configuration and will return success result

    bool res;
    // res = wm.autoConnect(); // auto generated AP name from chipid
    // res = wm.autoConnect("AutoConnectAP"); // anonymous ap
    res = wm.autoConnect("AutoConnectAP","password"); // password protected ap

    if(!res) {
        Serial.println("Failed to connect");
        // ESP.restart();
    } 
    else {
        //if you get here you have connected to the WiFi    
        Serial.println("connected...yeey :)");
    }

}

void loop() {
    // put your main code here, to run repeatedly:   
}

### Debug Messages

20:17:49.346 -> ESP-ROM:esp32c3-api1-20210207
20:17:49.346 -> Build:Feb  7 2021
20:17:49.346 -> rst:0x1 (POWERON),boot:0xd (SPI_FAST_FLASH_BOOT)
20:17:49.346 -> SPIWP:0xee
20:17:49.346 -> mode:DIO, clock div:2
[... 145 lines cut ...]
20:18:29.941 -> [ 40619][V][Parsing.cpp:226] _parseRequest(): headerName: Host
20:18:29.941 -> [ 40624][V][Parsing.cpp:227] _parseRequest(): headerValue: 192.168.4.1
20:18:29.941 -> [ 40631][V][Parsing.cpp:226] _parseRequest(): headerName: Connection
20:18:29.987 -> [ 40637][V][Parsing.cpp:227] _parseRequest(): headerValue: keep-alive
20:18:29.987 -> [ 40643][V][Parsing.cpp:226] _parseRequest(): headerName: Upgrade-Insecure-Requests
20:18:29.987 -> [ 40650][V][Parsing.cpp:227] _parseRequest(): headerValue: 1
20:18:29.987 -> [ 40656][V][Parsing.cpp:226] _parseRequest(): headerName: User-Agent
20:18:29.987 -> [ 40662][V][Parsing.cpp:227] _parseRequest(): headerValue: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/111.0.0.0 Safari/537.36
20:18:29.987 -> [ 40677][V][Parsing.cpp:226] _parseRequest(): headerName: Accept
20:18:30.034 -> [ 40682][V][Parsing.cpp:227] _parseRequest(): headerValue: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.7
20:18:30.034 -> [ 40699][V][Parsing.cpp:226] _parseRequest(): headerName: Referer
20:18:30.034 -> [ 40705][V][Parsing.cpp:227] _parseRequest(): headerValue: http://192.168.4.1/
20:18:30.034 -> [ 40712][V][Parsing.cpp:226] _parseRequest(): headerName: Accept-Encoding
20:18:30.034 -> [ 40718][V][Parsing.cpp:227] _parseRequest(): headerValue: gzip, deflate
20:18:30.034 -> [ 40725][V][Parsing.cpp:226] _parseRequest(): headerName: Accept-Language
20:18:30.081 -> [ 40731][V][Parsing.cpp:227] _parseRequest(): headerValue: it-IT,it;q=0.9,en-US;q=0.8,en;q=0.7,de;q=0.6
20:18:30.081 -> [ 40741][V][Parsing.cpp:254] _parseArguments(): args: 
20:18:30.081 -> [ 40746][V][Parsing.cpp:237] _parseRequest(): Request: /wifi
20:18:30.081 -> [ 40751][V][Parsing.cpp:238] _parseRequest():  Arguments: 
20:18:36.706 -> [ 47390][V][WiFiGeneric.cpp:383] _arduino_event_cb(): SCAN Done: ID: 129, Status: 0, Results: 5
20:18:36.706 -> [ 47390][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 1 - SCAN_DONE
20:18:36.706 -> *wm:5 networks found
20:18:36.751 -> [ 47414][V][WebServer.cpp:296] handleClient(): New client: client.localIP()=192.168.4.1

In this moment it disconnects

tablatronix, 2023-04-07:
This is usually related to a flood of requests. I usually see this on apple devices. You can try making sure all eth are disconnected on the device before connecting?

micdj, 2023-04-08:
All the eth on the esp? Yes esp Is in new flash state without any other connections

tablatronix, 2023-04-08:
No the client device

micdj, 2023-04-10:
No other connections on client, in different platforms