tzapu/WiFiManager #1583: [ERROR] WiFiManagerParameter is out of scope BUT it's not out of scope

olonsoft, 2023-04-08:
I was puzzled by this problem for one week until I found it was not my code's fault.

I kept getting `[ERROR] WiFiManagerParameter is out of scope` while I was trying to save my custom parameters.
I completely refactored my code many times, but the result was the same.

I could successfully load the parameters and display them while starting the webportal but when I was saving them, there was the mentioned error.

Debugging wm code I saw that in function 
```c++
void WiFiManager::doParamSave(){
...
if (_params[i] == NULL || _params[i]->_length == 0) {
```
the `_length` was always 0.

In a similar function `getParamOut()`  the `if`  is 

`if (_params[i] == NULL || _params[i]->_length > 99999) {`

So I changed the `doParamSave()` code to 

`if (_params[i] == NULL || _params[i]->_length > 99999) {`

and now my application works without a problem.

Using the latest version as of 2023-04-08

realA10001986, 2023-05-02:
That happens with customHTML-WiFiManagerParameter parameters, and is clearly a regression.

The _length field of such Parameters is now set to 0 (previously it was set to 1) in WiFiManagerParameter::WiFiManagerParameter(const char *custom), see line 34 of WiFiManager.cpp.

If _length 0 is now a thing, the OP is correct as to change the check.

closed 2023-05-02 by tablatronix

burner-, 2023-12-10:
Looks that this is closed at May 2 but not fiex in May 11 release. Advanced example ino wont work properly because of this bug.