tzapu/WiFiManager #1586: Can't build project in platformio

toxazhl, 2023-04-13:
#### Hardware
WiFimanager Branch/Release:  Master
Esp8266/Esp32: 
Hardware:			ESP32-S
Core Version:      	2.4.0, staging

### Description

I use VSCode and PlatformIO. I can't build a project for any framework (arduino, espidf)
My platformIO file:
```
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = espidf
upload_speed = 11500
lib_deps = https://github.com/tzapu/WiFiManager.git
```

and error that I get
```
Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: espidf)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (6.1.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf @ 3.50001.0 (5.0.1) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40500.0 (4.5.0) 
 - tool-ninja @ 1.9.0 
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0) 
 - toolchain-xtensa-esp32 @ 11.2.0+2022r1
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/esp32doit-devkit-v1/src/main.o
src/main.c:1:10: fatal error: WiFiManager.h: No such file or directory

*********************************************************************
* Looking for WiFiManager.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WiFiManager.h"
* Web  > https://registry.platformio.org/search?q=header:WiFiManager.h
*
*********************************************************************

    1 | #include <WiFiManager.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
Archiving .pio/build/esp32doit-devkit-v1/esp-idf/app_trace/libapp_trace.a
Indexing .pio/build/esp32doit-devkit-v1/esp-idf/app_trace/libapp_trace.a
Compiling .pio/build/esp32doit-devkit-v1/app_update/esp_ota_app_desc.o
Compiling .pio/build/esp32doit-devkit-v1/bootloader_support/src/bootloader_common.o
Compiling .pio/build/esp32doit-devkit-v1/bootloader_support/src/bootloader_common_loader.o
Compiling .pio/build/esp32doit-devkit-v1/bootloader_support/src/bootloader_clock_init.o
Compiling .pio/build/esp32doit-devkit-v1/bootloader_support/bootloader_flash/src/bootloader_flash.o
*** [.pio/build/esp32doit-devkit-v1/src/main.o] Error 1
Compiling .pio/build/esp32doit-devkit-v1/bootloader_support/src/bootloader_mem.o

```

I also noticed that library code is inactive after some condition
[image]

Is case if change framework = arduino in paltformio.ini file, I get this error

```Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (6.1.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20007.0 (2.0.7) 
 - tool-esptoolpy @ 1.40500.0 (4.5.0) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 34 compatible libraries
Scanning dependencies...
Dependency Graph
|-- WiFiManager @ 2.0.15-rc.1+sha.4611c75
|   |-- DNSServer @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |-- ESPmDNS @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |-- Update @ 2.0.0
|   |-- WebServer @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |-- WiFi @ 2.0.0
Building in release mode
Compiling .pio/build/esp32doit-devkit-v1/src/main.c.o
In file included from src/main.c:1:
.pio/libdeps/esp32doit-devkit-v1/WiFiManager/WiFiManager.h:23:10: fatal error: vector: No such file or directory
 #include <vector>
          ^~~~~~~~
compilation terminated.
Archiving .pio/build/esp32doit-devkit-v1/lib4e8/libWiFi.a
Archiving .pio/build/esp32doit-devkit-v1/libab2/libDNSServer.a
Archiving .pio/build/esp32doit-devkit-v1/libd0c/libESPmDNS.a
Indexing .pio/build/esp32doit-devkit-v1/libd0c/libESPmDNS.a
Indexing .pio/build/esp32doit-devkit-v1/lib4e8/libWiFi.a
Indexing .pio/build/esp32doit-devkit-v1/libab2/libDNSServer.a
Compiling .pio/build/esp32doit-devkit-v1/libaf7/Update/Updater.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/lib983/FS/FS.cpp.o
Compiling .pio/build/esp32doit-devkit-v1/lib983/FS/vfs_api.cpp.o
*** [.pio/build/esp32doit-devkit-v1/src/main.c.o] Error 1
Compiling .pio/build/esp32doit-devkit-v1/lib9d3/WebServer/Parsing.cpp.o
```

tablatronix, 2023-04-13:
/src/main.c

try c++ not c

dalbert2, 2023-04-14:
It definitely builds in PlatformIO; I am using it there.

closed 2023-04-15 by tablatronix