tzapu/WiFiManager #1587: Using PROGMEM for string arrays

dalbert2, 2023-04-14:
### Basic Infos
string arrays in wm_consts_en.h aren't being placed in PROGMEM properly
#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32: ESP8266

Hardware:			ESP-12e

Core Version:      	3.0.5

### Description

Moving strings into PROGMEM is essential for ESP8266 where IRAM is extremely tight.  

This code is incorrect (see wm_consts_en.h):
```
const char * const WIFI_STA_STATUS[] PROGMEM
{
  "WL_IDLE_STATUS",     // 0 STATION_IDLE
  "WL_NO_SSID_AVAIL",   // 1 STATION_NO_AP_FOUND
  "WL_SCAN_COMPLETED",  // 2
  "WL_CONNECTED",       // 3 STATION_GOT_IP
  "WL_CONNECT_FAILED",  // 4 STATION_CONNECT_FAIL, STATION_WRONG_PASSWORD(NI)
  "WL_CONNECTION_LOST", // 5
  "WL_DISCONNECTED",    // 6
  "WL_STATION_WRONG_PASSWORD" // 7 KLUDGE
};
```
The above code places the pointers to the string constants in PROGMEM, but the string constants themselves land in RAM.
The below code shows how to put both the pointers and the strings themselves in PROGMEM:
```
/** TCP state descriptions */
static const char tcp_closed_str[]     PROGMEM = "CLOSED";
static const char tcp_listen_str[]     PROGMEM = "LISTEN";
static const char tcp_syn_sent_str[]   PROGMEM = "SYN_SENT";
static const char tcp_syn_rcvd_str[]   PROGMEM = "SYN_RCVD";
static const char tcp_fin_wait_1_str[] PROGMEM = "FIN_WAIT_1";
static const char tcp_fin_wait_2_str[] PROGMEM = "FIN_WAIT_2";
static const char tcp_close_wait_str[] PROGMEM = "CLOSE_WAIT";
static const char tcp_closing_str[]    PROGMEM = "CLOSING";
static const char tcp_last_ack_str[]   PROGMEM = "LAST_ACK";
static const char tcp_time_wait_str[]  PROGMEM = "TIME_WAIT";
PGM_P tcp_state_str[] PROGMEM = {
     tcp_closed_str, tcp_listen_str, tcp_syn_sent_str, tcp_syn_rcvd_str, tcp_fin_wait_1_str,
     tcp_fin_wait_2_str, tcp_close_wait_str, tcp_closing_str, tcp_last_ack_str, tcp_time_wait_str };
```
I know this is ugly, but it works; if someone knows a prettier solution, please share it.  At present, WiFiManager uses 1344 of IRAM even if you don't instantiate an instance of the class which is a problem for many applications on the ESP8266.

### Settings in IDE
### Sketch
### Debug Messages

dalbert2, 2023-04-14:
Even worse, the string constants are all declared in header files which are included by WiFiManager.h so any class or module that includes WiFiManager.h creates a copy of all of the string constants.  Why are these not defined in a compilation unit (.c or .cpp)?

tablatronix, 2023-04-15:
What am I missing here? I am not seeing this memory waste

```php
RAM:   [====      ]  40.0% (used 32732 bytes from 81920 bytes)
Flash: [====      ]  36.9% (used 385916 bytes from 1044464 bytes)
.pio/build/nodemcuv2/firmware.elf  :
section            size         addr
.data              1612   1073643520
.noinit              60   1073645132
.text               463   1074790400
.irom0.text      351480   1075843088
.text1            27993   1074790864
.rodata            4368   1073645200
.bss              26752   1073649568
.comment           5066            0
.xtensa.info         56            0                                                                                                                                                                              56            0

total 1366237

------------------------------------
BEFORE

RAM:   [====      ]  40.0% (used 32732 bytes from 81920 bytes)
Flash: [====      ]  36.9% (used 385836 bytes from 1044464 bytes)
.pio/build/nodemcuv2/firmware.elf  :
section            size         addr
.data              1612   1073643520
.noinit              60   1073645132
.text               463   1074790400
.irom0.text      351400   1075843088
.text1            27993   1074790864
.rodata            4368   1073645200
.bss              26752   1073649568
.comment           5066            0
.xtensa.info         56            0

total 1366157
```

dalbert2, 2023-04-15:
@tablatronix please see: https://arduino-esp8266.readthedocs.io/en/latest/PROGMEM.html#how-do-i-declare-arrays-of-strings-in-progmem-and-retrieve-an-element-from-it

If you do this with the string arrays, you will see free heap increase.

I also think the strings need to be moved into a separate compilation unit (e.g. wm_strings.cpp).  You can still use the pre-processor to select a language and a header file to define extern references for strings needed by other compilation units.  Example:

strings.h
```
#pragma once

#define LANG_EN 0
#define LANG_ES 1

extern const char *_str_table[];
```

strings.c
```
#include "strings.h"

#ifndef LANG
   #define LANG LANG_EN
#endif

#if (LANG == LANG_EN)
    static const char _hello[] = "hello";
    static const char _world[] = "world";
#elif (LANG == LANG_ES)
    static const char _hello[] = "hola";
    static const char _world[] = "mundo";
#endif

const char *_str_table[] = { _hello, _world };
```
main.c:
```
#include <stdio.h>

#include "strings.h"

int main(int argc, char **argv) {
    printf("%s %s\n", _str_table[0], _str_table[1]);
    return 0;
}
```
Compile for english:gcc strings.c main.c
Compile for spanish: gcc strings.c main.c -DLANG=LANG_ES

timr49, 2025-07-31:
> What am I missing here? I am not seeing this memory waste
> ...

I checked the contents of the compiled code in the .o, .elf and .bin files using strings(1).
When two cpp files (typically WiFiManager.cpp and the calling code)  include WiFiManager.h and hence wm_strings_en.h
there are indeed multiple copies of each of the flash memory strings compiled into the .o files and linked into the .elf file. However the process that converts the .elf file to the .bin file deduplicates them such that the final executable does *not* contain duplicates.

I checked this with both of Arduino IDE ESP8266 3.1.2 and ESP32 3.2.0.

Per @dalbert2, usual practice would be to put only declarations in a .h file and definitions in a .cpp file. In this case, only WiFiManager.cpp needs to reference these strings, so that could be done simply by including the strings file in WiFiManager.cpp rather than WiFiManager.h and moving the few initialisations that use flash strings in WiFiManager.h from there to WiFiManagerInit() in WiFiManager.cpp.  As shown above, it doesn't buy us any memory, though it would decrease compile times by some unknown amount.

tablatronix, 2025-07-31:
Something is wrong with the compiler if its ignoring include guards, perhaps include in the main source file and see if that changes it.

timr49, 2025-07-31:
> Something is wrong with the compiler if its ignoring include guards, perhaps include in the main source file and see if that changes it.

I disagree. Include guards only ever stop one .cpp file from including the same .h file more than once. They cannot stop multiple .cpp files from each including the same .h file once each, which is what is happening.

As suggested, including the strings file in WiFiManager.cpp rather than WiFiManager.h is preferable.