tzapu/WiFiManager #1588: Example for making connecting to WLAN robust

probonopd, 2023-04-16:
Often one finds this

```
#include <WiFiManager.h>

void setup() {
  WiFiManager wifiManager;
  wifiManager.autoConnect();
}

void loop() {
  // main program logic here
}
```

This has two downsides:
* If the WLAN connection drops while the device is connected, it never re-connects when the WLAN re-appears
* If the WLAN is not in range while the device is powered on, then the device will hang on the configuration portal forever

To make it more robust, I would suggest an approach that:
* If the WLAN is not in range or is not configured, opens the configuration portal only for 180 seconds
* After the timeout, continously tries to reconnect to the WLAN
* If the WLAN is in range when the device is powered on but goes out of range thereafter, then the device should not open the configuration portal forever but should periodically try to join the network again

Is the following code suitable to do that?

```
#include <WiFiManager.h> 

WiFiManager wifiManager;

void setup() {
  wifiManager.setConnectTimeout(60); // Set the connection timeout to 60 seconds
  wifiManager.setConfigPortalTimeout(180); // Set the configuration portal timeout to 180 seconds
  wifiManager.autoConnect(); // Try to connect to the WLAN with saved credentials, or create an AP with a default name if not successful for 180 seconds
}

void loop() {
  if (WiFi.status() != WL_CONNECTED) { // Check if the device is not connected to the WLAN
    Serial.println("Failed to connect to the WLAN");
    delay(5000); // Wait 5 seconds before trying again
    wifiManager.setAutoConnect(false); // Disable the autoconnect feature of WiFiManager so that the portal is not shown again
    wifiManager.connect(); // Try to connect to the WLAN again
    return; // Return from the loop to allow the device to reconnect to the WLAN
  }
  
   // main program logic here
}

```

I suggest to add such an example to the documentation.

probonopd, 2023-04-16:
Is the following statement correct?

The configuration portal timeout of 180 seconds in the example code refers to the time that the library will wait for the user to connect to the access point and open the configuration portal. Once the user has connected to the access point and opened the configuration portal, they can take as much time as they need to enter the WiFi credentials and save them (not just 180 seconds).

Laxilef, 2023-09-15:
I also see this problem

B1ackt34, 2025-02-15:
Thank you for the example, a good one to start from.

closed 2025-02-16 by tablatronix

probonopd, 2025-02-17:
> [tablatronix](https://github.com/tablatronix) closed this as [completed](https://github.com/tzapu/WiFiManager/issues?q=is%3Aissue%20state%3Aclosed%20archived%3Afalse%20reason%3Acompleted)[2 days ago](https://github.com/tzapu/WiFiManager/issues/1588#event-16323203091)

Was this implemented?

tablatronix, 2025-02-18:
Its in user code, not something we need to add to wm core