tzapu/WiFiManager #1590: Example for custom parameters

probonopd, 2023-04-16:
Documentation says

> You can use WiFiManager to collect more parameters than just SSID and password. This could be helpful for configuring stuff like MQTT host and port, [blynk](http://www.blynk.cc/) or [emoncms](http://emoncms.org/) tokens, just to name a few. You are responsible for saving and loading these custom values. The library just collects and displays the data for you as a convenience. 

I think there should be an example showing how to actually do this. Something like this:

```
#include <WiFiManager.h>
#include <Preferences.h>

char mqtt_server[40];
int mqtt_port;
char mqtt_username[40];
char mqtt_password[40];
char mqtt_topic[40] = "your_topic";

Preferences preferences;

void saveConfigCallback() {
  preferences.putString("mqtt_server", mqtt_server);
  preferences.putInt("mqtt_port", mqtt_port);
  preferences.putString("mqtt_username", mqtt_username);
  preferences.putString("mqtt_password", mqtt_password);
  preferences.putString("mqtt_topic", mqtt_topic);
}

void setup() {

  WiFiManager wifiManager;
  wifiManager.setSaveConfigCallback(saveConfigCallback);
  wifiManager.setConfigPortalTimeout(180);

  preferences.begin("mqtt_config", false);
  strcpy(mqtt_server, preferences.getString("mqtt_server", "default_mqtt_server").c_str());
  mqtt_port = preferences.getInt("mqtt_port", 1883);
  strcpy(mqtt_username, preferences.getString("mqtt_username", "default_username").c_str());
  strcpy(mqtt_password, preferences.getString("mqtt_password", "default_password").c_str());
  strcpy(mqtt_topic, preferences.getString("mqtt_topic", "default_mqtt_topic").c_str());

  WiFiManagerParameter custom_mqtt_server("server", "MQTT Server", mqtt_server, 40);
  WiFiManagerParameter custom_mqtt_port("port", "MQTT Port", String(mqtt_port).c_str(), 6);
  WiFiManagerParameter custom_mqtt_username("username", "MQTT Username", mqtt_username, 40);
  WiFiManagerParameter custom_mqtt_password("password", "MQTT Password", mqtt_password, 40);
  WiFiManagerParameter custom_mqtt_topic("topic", "MQTT Topic", mqtt_topic, 40);

  wifiManager.addParameter(&custom_mqtt_server);
  wifiManager.addParameter(&custom_mqtt_port);
  wifiManager.addParameter(&custom_mqtt_username);
  wifiManager.addParameter(&custom_mqtt_password);
  wifiManager.addParameter(&custom_mqtt_topic);

  wifiManager.setCustomHeadElement("<style>body{max-width:500px;margin:auto}</style>");

  wifiManager.autoConnect();

  strcpy(mqtt_server, custom_mqtt_server.getValue());
  mqtt_port = atoi(custom_mqtt_port.getValue());
  strcpy(mqtt_username, custom_mqtt_username.getValue());
  strcpy(mqtt_password, custom_mqtt_password.getValue());
  strcpy(mqtt_topic, custom_mqtt_topic.getValue());

  preferences.putString("mqtt_server", mqtt_server);
  preferences.putInt("mqtt_port", mqtt_port);
  preferences.putString("mqtt_username", mqtt_username);
  preferences.putString("mqtt_password", mqtt_password);
  preferences.putString("mqtt_topic", mqtt_topic);

  // Now can use the variables, e.g., like this:
  // mqttClient.setServer(mqtt_server, mqtt_port);
}

void loop() {

}
```

Halvhjearne, 2023-04-27:
this was super usefull but.
however i could be wrong but i think i think the way you do it there, the device will write to the flash on every reboot if im not mistaken.
maybe it would be better to have the saveConfigCallback change a bool to true and then put line 55-59 inside an if statement that triggers from said bool?

closed 2023-04-27 by tablatronix

probonopd, 2023-04-27:
Where? I can't seem to find them on https://github.com/tzapu/WiFiManager/wiki/Examples.

probonopd, 2023-04-29:
Thanks, maybe it should be linked from https://github.com/tzapu/WiFiManager/wiki/Examples.

tablatronix, 2023-05-31:
Sorry I didn't even see the example, I will take a look. I personally use structs and just save/read them from flash or wherever.

tablatronix, 2023-06-12:
I have never heard of it, thanks.

jonathanendersby, 2023-08-05:
@tzapu  These examples should be in the actual repo. They are clean and work!