tzapu/WiFiManager #1600: Updating WiFiManager in Arduino IDE vers. 1.8.19

alaintanguy, 2023-05-05:
## PLEASE TRY Latest Master BRANCH before submitting bugs, in case they were already fixed. ##

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware ESP32 S2 Adafruit Feather
WiFimanager Branch/Release:  Master

Esp8266/Esp32: ESP32

Hardware:			ESP-12e, esp01, esp25

Core Version:      	2.4.0, staging

### Description

Problem description

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries with Libraries Manager in Arduino IDE 1.8.19: 
Trying to update library WiFiManger from version 2.0.11 Beta to  2.0.15 Rc-1 :
 Click on Select version - Select 2.0.15 Rc-1 - 
Press Install Shows message: Updating list of installed lib On bottom window
Seems to do something but at the end the version still at 2.0.11
### Sketch

```cpp
#BEGIN
#include <Arduino.h>

void setup() {

}

void loop() {

}
#END
```

### Debug Messages

```
messages here
```

alaintanguy, 2023-05-05:
Was able to install last version with zip file that changed the version on lib. Manager

tablatronix, 2023-05-05:
Not sure what you mean

alaintanguy, 2023-05-06:
I was trying to update WifiManager directly from the Arduino IDE but, for
some reason, the library manager did not change the version after I tried
it. I finally was able to see the changed version after I downloaded and
installed the zip WManager file. Now everything are go ;-)
Thanks for your message.
Alain Samoun