tzapu/WiFiManager #1601: Proble ubdating Custom parameters

ahmetcemturan, 2023-05-10:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32:  Wemos D1 Mini (ESP8266)

Hardware:			ESP-12e

Core Version:      	2.4.0, staging

### Description
I am building a weather station and I have an extensive set of custom parameters for this, as each website needs other credentials and passwords.
If I first enter all of the custom parameters and the setup wifi, it will save and all is ok.
If I leave some parameters out or need to change them later, it wont save them.  Only if I also update wifi, are the new parameters saved.

### Settings in IDE

Module:  			 Wemos D1

Additional libraries: 
#include <LittleFS.h>
#include <Arduino.h>

#include <ESP8266WiFi.h>
#include "PubSubClient.h" //Nick O'Leary   http://knolleary.net
#include <Ticker.h>
#include <ESP8266WebServer.h>
#include <ESP8266HTTPClient.h>
#include <WiFiClientSecure.h>
#include <ESP8266httpUpdate.h>
#include "WiFiManager.h" //https://github.com/tzapu/WiFiManager
#include <DNSServer.h>
#include <ArduinoJson.h> //https://github.com/bblanchon/ArduinoJson
#include "TimeLib.h" // https://github.com/michaelmargolis/arduino_time by Michael Margolis
#include "AS5600.h"//  https://github.com/Seeed-Studio/Seeed_Arduino_AS5600
#include "BME280I2C.h"//  https://github.com/finitespace/BME280
#include <Wire.h>
#include "runningAngle.h"//https://github.com/RobTillaart/runningAngle
#include <SPI.h>
					 

### Sketch

```cpp
#BEGIN
#include <Arduino.h>

void setup() {

}

void loop() {

}
#END
```

### Debug Messages

```
messages here
```

ahmetcemturan, 2023-05-10:
I just saw some posts about core versions.  Should I downgrade it?

ahmetcemturan, 2023-05-10:
My current workaround is:
Set teh new parameters,
Save
Go To Wifi settings
just enter some non existent SSID
Save.   İt will obviously fail to connect
Turn On the Wifi You really want to connect to and enter SSID and Password.
Save.
It will connect and then save parameters..

tablatronix, 2023-05-11:
#1558

ahmetcemturan, 2023-05-11:
Yes it is.  Same here.  Fooled by the existing "pre filled in data".
Worsened by the fact that I have to turn off that acces point to be able to access the WifiManager portal (or is there an other way)
Also is it not  the normal expectation that when You click Save on the parameters page that the parameters are already saved so that when you restart (either by portal or manually) the settings entered are saved?
Just my thoughts..
[image]

tablatronix, 2023-05-31:
Are you all sure you are using the correct event handlers for saves or return values?

I cannot reproduce this