tzapu/WiFiManager #1602: AP mode if saved network not available

micronthings, 2023-05-11:
After device started, AP mode starts, if saved network is not available. Is there any way to turn this feature off?
My problem is: in case of electrical blackout the device can't connect if the router starts later than the device. (WiFi network is not exist at the moment when device starts). For my use case would better if the device tries connect to saved network permanently in this case.

toome123, 2023-05-18:
i think you can just restart the device if no connection established. Here example with ESP 
`  bool isConnected = wifiManager.autoConnect("WIFI", "password");  
  if(isConnected){
    Serial.println("Connected to WIFI");
  }else{
    Serial.println("Unable to connect to WIFI");
    Serial.println("failed to connect and hit timeout");
        delay(3000);
        //reset and try again, or maybe put it to deep sleep
        ESP.restart();
        delay(5000);`
  }

tablatronix, 2023-05-18:
You can use the configportaltimeout or add some checks to see if creds are saved or not or use ondemand only. This is asked many times here in issues