tzapu/WiFiManager #1603: ESP32 Watchdog timeout triggered repeatedly

chriskinal, 2023-05-11:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  2.0.15-rc.1

Esp32: 

Hardware:			ESP32-WROOM-DA Module

Core Version:      	

### Description

Problem description

Upload "Basic" example to an ESP32.
Start the system and connect a serial terminal.
Connect Windows laptop to "AutoconnectAP"
Redirected to WiFi Manager portal.
Click on an AP from the list of scanned AP's.
Type the pasword
A watchdog time out is triggered and reboots the system.

### Settings in IDE

Module:  			ESP32-WROOM-DA Module

Additional libraries: 

### Sketch

#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

void setup() {
    // WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP
    // it is a good practice to make sure your code sets wifi mode how you want it.

    // put your setup code here, to run once:
    Serial.begin(115200);
    
    //WiFiManager, Local intialization. Once its business is done, there is no need to keep it around
    WiFiManager wm;

    // reset settings - wipe stored credentials for testing
    // these are stored by the esp library
    // wm.resetSettings();

    // Automatically connect using saved credentials,
    // if connection fails, it starts an access point with the specified name ( "AutoConnectAP"),
    // if empty will auto generate SSID, if password is blank it will be anonymous AP (wm.autoConnect())
    // then goes into a blocking loop awaiting configuration and will return success result
    wm.setDebugOutput(true);
    bool res;
    // res = wm.autoConnect(); // auto generated AP name from chipid
    // res = wm.autoConnect("AutoConnectAP"); // anonymous ap
    res = wm.autoConnect("AutoConnectAP","password"); // password protected ap

    if(!res) {
        Serial.println("Failed to connect");
        // ESP.restart();
    } 
    else {
        //if you get here you have connected to the WiFi    
        Serial.println("connected...yeey :)");
    }

}

void loop() {
    // put your main code here, to run repeatedly:   
}

### Debug Messages

06:13:12.936 -> ets Jul 29 2019 12:21:46
06:13:12.936 -> 
06:13:12.936 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
06:13:12.936 -> configsip: 0, SPIWP:0xee
06:13:12.936 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
06:13:12.936 -> mode:DIO, clock div:1
06:13:12.936 -> load:0x3fff0030,len:1344
06:13:12.936 -> load:0x40078000,len:13924
06:13:12.936 -> ho 0 tail 12 room 4
06:13:12.936 -> load:0x40080400,len:3600
06:13:12.969 -> entry 0x400805f0
06:13:13.266 -> *wm:AutoConnect 
06:13:13.366 -> *wm:No wifi saved, skipping 
06:13:13.366 -> *wm:AutoConnect: FAILED 
06:13:13.366 -> *wm:StartAP with SSID:  AutoConnectAP
06:13:13.856 -> *wm:AP IP address: 192.168.4.1
06:13:13.856 -> *wm:Starting Web Portal 
06:13:59.939 -> *wm:9 networks found
06:14:04.946 -> E (51925) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
06:14:04.946 -> E (51925) task_wdt:  - IDLE (CPU 0)
06:14:04.946 -> E (51925) task_wdt: Tasks currently running:
06:14:04.946 -> E (51925) task_wdt: CPU 0: loopTask
06:14:04.946 -> E (51925) task_wdt: CPU 1: IDLE
06:14:04.946 -> E (51925) task_wdt: Aborting.
06:14:04.946 -> 
06:14:04.980 -> abort() was called at PC 0x400ec529 on core 0
06:14:04.980 -> 
06:14:04.980 -> 
06:14:04.980 -> Backtrace: 0x40083641:0x3ffbec7c |<-CORRUPTED
06:14:04.980 -> 
06:14:04.980 -> 
06:14:04.980 -> 
06:14:04.980 -> 
06:14:04.980 -> ELF file SHA256: 576a586d4c46321f
06:14:04.980 -> 
06:14:05.143 -> Rebooting...
06:14:05.143 -> ets Jul 29 2019 12:21:46
06:14:05.143 -> 
06:14:05.143 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
06:14:05.143 -> configsip: 0, SPIWP:0xee
06:14:05.143 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
06:14:05.177 -> mode:DIO, clock div:1
06:14:05.177 -> load:0x3fff0030,len:1344
06:14:05.177 -> load:0x40078000,len:13924
06:14:05.177 -> ho 0 tail 12 room 4
06:14:05.177 -> load:0x40080400,len:3600
06:14:05.177 -> entry 0x400805f0
06:14:05.472 -> *wm:AutoConnect 
06:14:05.570 -> *wm:No wifi saved, skipping 
06:14:05.570 -> *wm:AutoConnect: FAILED 
06:14:05.604 -> *wm:StartAP with SSID:  AutoConnectAP
06:14:06.095 -> *wm:AP IP address: 192.168.4.1
06:14:06.095 -> *wm:Starting Web Portal

chriskinal, 2023-05-11:
I switched the Arduino IDE from ESP32-WROOM-DA module to ESP32 Dev Module and added some lines to reset the watchdog within every 'while' loop.

#include "esp_task_wdt.h"

 Example:

  while(retry <= _connectRetries && (connRes!=WL_CONNECTED)){
  esp_task_wdt_reset();
.....rest of code

Just switching the board or just adding watchdog reset did not resolve the issue. I had to do both.

tablatronix, 2023-05-11:
interesting, did you try to just add a delay() in the while loop ?

chriskinal, 2023-05-12:
I tried delay() which did not help. I think because it creates some sort of tight loop that only makes the problem worse.

I tried yield() and it seems to be working. I put one in all 5 "while loops" in the code. I found that not having them in all "while loops" caused the issue when execution hit that part of the function.

tablatronix, 2023-05-12:
That is the only loop without a delay or yield

chriskinal, 2023-05-12:
I found 5. Lines 295, 786, 1041, 1250 & 1540. 295 has a delay() but that did not seem to work.

tablatronix, 2023-05-12:
`06:14:04.980 -> Backtrace: 0x40083641:0x3ffbec7c |<-CORRUPTED`

I would love to see a decoded backtrace for this, I wonder where its failing

chriskinal, 2023-05-12:
CPU = ESP32-WROOM-32 
Freq = 240MHz
Erase flash before upload.
Flash freq = 80MHz
Flash mode = QIO
Upload speed = 921600

I am using a generic ESP32 board that has the above CPU on it. I have used the ESP32-WROOM-DA board profile in the Arduino IDE with this board on a number of projects and never had an issue. I can reliably reproduce the issue with the "Basic" example in the WiFiManager package. The only difference I noticed is when using the ESP32-WROOM-DA board profile is used the blue wifi led on the board flashes with wifi activity. Using the ESP32 Dev Module profile the blue wifi led is off. The blue led is on GPIO 2.

tablatronix, 2023-05-12:
Have you tried 2.0.16 yet?

chriskinal, 2023-05-12:
I am using 2.1.0 (latest) as I am on a Mac.

tablatronix, 2023-05-12:
? I mean WM ?

chriskinal, 2023-05-12:
Ooops. I thought you meant Arduino IDE version. wm is at 2.0.16-rc.2.

chriskinal, 2023-05-19:
So I tested this with multiple ESP32 board profiles in Arduino IDE. It seems any profile I tried that makes use of the blue LED (GPIO2) triggers the watchdog issue. Board profiles that do not make use of the blue LED function as expected. I have a solution for my project by using a board profile that does not use the blue LED. The board is inside an enclosure so I can't see the blue light even if it was working. ;-)

Up to the devs on how they want to close out this issue.

tablatronix, 2023-05-19:
curious, Do you have setDebugOutput(true); in your code?

chriskinal, 2023-05-19:
I am using the Basic example from the WM repo with no modifications. setDebugOutput() is not set in the example.

Not really any codebase to check other than WM.

tablatronix, 2023-05-19:
Ok let me try to test one of those board defs

chriskinal, 2023-05-19:
Here are the board manger URL's I use.

https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/arduino/package_m5stack_index.json
https://www.pjrc.com/teensy/package_teensy_index.json

tablatronix, 2023-05-31:
Anything else on this? I do not have a DA module, I will try using DA def on a regular ESP32 but i doubt it will reproduce, also i do not know which esp lib ver

chriskinal, 2023-05-31:
I feel like it is something upstream in the board definition files. Changing to a board definition file for a board that does not have the blue LED works fine. Adding Yield() to the loops in WM works fine. I can do digitalWrite(..) to the GPIO to manipulate the blue LED fine. 

Thanks for all your help. Please feel free to close this one one out since I have multiple work-arounds and it feels like an upstream problem.

tablatronix, 2023-05-31:
Sounds like a bad pin def file

tablatronix, 2023-06-01:
What dev board are you using?

chriskinal, 2023-06-01:
A no-name board from Amazon. The chip says ESP32-WROOM-32. The underside says ESP32S. It is of the 30 pin variety.

tablatronix, 2023-06-01:
Then why are you using a DA profile?

chriskinal, 2023-06-01:
It was one of two in Arduino IDS that were related to the WROOM. the uPesy ESP32 WROOM DevKit profile did the same thing.

dagi0627, 2023-12-20:
#include "Arduino.h"
#include "WiFiManager.h"

 WiFiManager wm;

extern "C" void app_main()
{
    initArduino();
    pinMode(2, OUTPUT);
    digitalWrite(2, HIGH);
    bool res;
    res = wm.autoConnect("OK-AP"); // anonymous ap
    if(!res) {
        Serial.println("Failed to connect");
        // ESP.restart();
    } 
    else {    
        Serial.println("connected...:)");
    }
    
}
But have erros what does it? How to fix it                                                                                                                                                                                                                                                                                               
 E (6373) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (6373) task_wdt:  - IDLE (CPU 0)
E (6373) task_wdt: Tasks currently running:
E (6373) task_wdt: CPU 0: main
E (6373) task_wdt: CPU 1: IDLE
E (6373) task_wdt: Print CPU 0 (current core) backtrace
Backtrace: 0x400F0607:0x3FFB1C60 0x400F078E:0x3FFB1C80 0x40082D4D:0x3FFB1CA0 0x400F4BD5:0x3FFBA790 0x400EB32A:0x3FFBA7F0 0x400E996A:0x3FFBA830 0x400E20DA:0x3FFBA890 0x400E246A:0x3FFBA8F0 0x400E2705:0x3FFBA9A0 0x400D7E88:0x3FFBAA70 0x4016D60F:0x3FFBAA90 0x4008B6E6:0x3FFBAAC0
0x400f0607: task_wdt_timeout_handling at /home/enhee/esp/esp-idf/components/esp_system/task_wdt/task_wdt.c:461 (discriminator 3)
0x400f078e: task_wdt_isr at /home/enhee/esp/esp-idf/components/esp_system/task_wdt/task_wdt.c:585
0x40082d4d: _xt_lowint1 at /home/enhee/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1226
0x400f4bd5: tryget_socket at /home/enhee/esp/esp-idf/components/lwip/lwip/src/api/sockets.c:474
0x400eb32a: WiFiServer::accept() at /home/enhee/regulator/wifiManager/components/arduino/libraries/WiFi/src/WiFiServer.cpp:57
0x400e996a: WebServer::handleClient() at /home/enhee/regulator/wifiManager/components/arduino/libraries/WebServer/src/WebServer.cpp:278
0x400e20da: WiFiManager::processConfigPortal() at /home/enhee/regulator/wifiManager/components/arduino/libraries/WiFiManager/WiFiManager.cpp:871
0x400e246a: WiFiManager::startConfigPortal(char const*, char const*) at /home/enhee/regulator/wifiManager/components/arduino/libraries/WiFiManager/WiFiManager.cpp:801
0x400e2705: WiFiManager::autoConnect(char const*, char const*) at /home/enhee/regulator/wifiManager/components/arduino/libraries/WiFiManager/WiFiManager.cpp:387
0x400d7e88: app_main at /home/enhee/regulator/wifiManager/main/main.cpp:12
0x4016d60f: main_task at /home/enhee/esp/esp-idf/components/freertos/app_startup.c:208 (discriminator 13)
0x4008b6e6: vPortTaskWrapper at /home/enhee/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162 
I am using esp-idf extension on vscode and Arduino ESP32 as ESP-IDF components

UradaSources, 2024-07-09:
i faced the same situation. A project using arduino, wifimanager as idf components on esp32c3. The arduino and wifimanager versions were the latest versions pulled directly from the repositories.
this error always occurs regardless of blocking/non-blocking, or using autoConnect or startConfigPortal.
these warnings don't seem to be of any use. They don't reboot the chip, but they pop up every second or less, which is very annoying.
```c++
extern "C" void app_main(void)
{
    initArduino();

    log_i("start2");

    WiFiManager wm;
    // wm.setConfigPortalTimeout(120);
    // wm.setConnectTimeout(10);

    // wm.autoConnect("");

    wm.startConfigPortal();
    wm.setBreakAfterConfig(false);

    while(wm.process()) delay(10);
    
    wm.stopConfigPortal();

    log_i("wifi: %s", WiFi.isConnected() ? "t" : "f");
    log_i("ip: %s", WiFi.localIP().toString().c_str());

    while(true) delay(1000);
}
```

log:
```
I (564) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
[   581][W][STA.cpp:537] disconnect(): STA already disconnected.
I (594) wifi:mode : softAP (dc:da:0c:f5:57:19)
E (6084) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (6084) task_wdt:  - IDLE (CPU 0)
E (6084) task_wdt: Tasks currently running:
E (6084) task_wdt: CPU 0: main
E (6084) task_wdt: Print CPU 0 (current core) registers
Core �2 register dump:
MEPC    : 0x42017688  RA      : 0x4201765c  SP      : 0x3fc9c890  GP      : 0x3fc92e00  
0x42017688: NetworkClient::connected() at D:/Espressif/global_components/arduino/libraries/Network/src/NetworkClient.cpp:574
0x4201765c: NetworkClient::connected() at D:/Espressif/global_components/arduino/libraries/Network/src/NetworkClient.cpp:541

TP      : 0x3fc727d0  T0      : 0x4005890e  T1      : 0xaa800000  T2      : 0xffffffff
0x4005890e: _data_end_btdm_rom in ROM

S0/FP   : 0x3fca87d4  S1      : 0x3fca87d4  A0      : 0xffffffff  A1      : 0x3fc9c8e8
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x3fc9a000  A5      : 0x00000000
A6      : 0xfa000000  A7      : 0x00000003  S2      : 0x3fc9e7f8  S3      : 0x3fc9cb68
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x0000001c  MTVEC   : 0x4005890e  MCAUSE  : 0x005cd890  MTVAL   : 0x403885de
0x4005890e: _data_end_btdm_rom in ROM
0x403885de: vTaskDelay at D:/Espressif/frameworks/esp-idf-v5.1.4/components/freertos/FreeRTOS-Kernel/tasks.c:1688

MHARTID : 0x3fc9ca4c
E (11084) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (11084) task_wdt:  - IDLE (CPU 0)
E (11084) task_wdt: Tasks currently running:
E (11084) task_wdt: CPU 0: main
E (11084) task_wdt: Print CPU 0 (current core) registers
Core  0 register dump:
MEPC    : 0x4201ad9c  RA      : 0x4201ad96  SP      : 0x3fc9c8c0  GP      : 0x3fc92e00  
0x4201ad9c: NetworkClient::operator bool() at D:/Espressif/global_components/arduino/libraries/Network/src/NetworkClient.h:75
 (inlined by) WebServer::handleClient() at D:/Espressif/global_components/arduino/libraries/WebServer/src/WebServer.cpp:411
0x4201ad96: NetworkClient::operator bool() at D:/Espressif/global_components/arduino/libraries/Network/src/NetworkClient.h:75
 (inlined by) WebServer::handleClient() at D:/Espressif/global_components/arduino/libraries/WebServer/src/WebServer.cpp:411

TP      : 0x3fc727d0  T0      : 0x4005890e  T1      : 0xaa800000  T2      : 0xffffffff
0x4005890e: _data_end_btdm_rom in ROM

S0/FP   : 0x3fca876c  S1      : 0x3fca87d4  A0      : 0x3fca87d4  A1      : 0x3fc9c8e8
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x3fc9a000  A5      : 0x42017648
0x42017648: NetworkClient::connected() at D:/Espressif/global_components/arduino/libraries/Network/src/NetworkClient.cpp:540

A6      : 0xfa000000  A7      : 0x00000003  S2      : 0x3fc9e7f8  S3      : 0x3fc9cb68
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00000000  MTVEC   : 0x00000000  MCAUSE  : 0x00000000  MTVAL   : 0x3c0cad84
MHARTID : 0x00000000
```

UradaSources, 2024-07-09:
this problem can be solved by using a non-blocking version
for the main function of idf style:
```c++

#define SENCOOND2MS(sencond) (std::uint64_t)(sencond * 1000);

extern "C" void app_main(void)
{
    initArduino();

    log_i("start2");

    WiFiManager wm;
    wm.setConnectTimeout(10);

    pinMode(4, OUTPUT); // debug

    wm.setConfigPortalBlocking(false);
    wm.startConfigPortal();

    log_i("wait");

    auto timeout = millis() + SENCOOND2MS(120);
    while (wm.process() == false && millis() < timeout)
    {
        digitalWrite(4, !digitalRead(4)); // debug
        delay(100);
    }

    if (wm.getConfigPortalActive())
        wm.stopConfigPortal();

    log_i("wifi: %s", WiFi.isConnected() ? "t" : "f");
    log_i("ip: %s", WiFi.localIP().toString().c_str());

    while(true)
    {
        // ...
        delay(1000);
    }
}
```