tzapu/WiFiManager #1605: How do I redirect the web browser after making changes in web portal?

nospam2k, 2023-05-11:
When I make changes from Web Portal (not as AP but via wm.startWebPortal), how can I redirect the web browser away from /wifi? back to /.

tablatronix, 2023-05-12:
Not sure I fully understand the question, are you only changing params, or saving wifi ?

nospam2k, 2023-05-13:
When I run the portal and check using wm.process() and then go to the URL /, I get the normal AP menu, but now as a web portal. When I go into wifi and just click save (I do have custom parameters in the wifi menu), I get back the page that just says SAVED. I want it to redirect back to / after like 5 seconds or so.

Halvhjearne, 2023-05-21:
@nospam2k idk if there is a better or prefered way to do this, but what i did was redirect back to main page, once you hit the saved message page, by editing this line in wm_strings_en.h:
`const char HTTP_PARAMSAVED[]       PROGMEM = "<div class='msg S'>Saved<br/></div>";`
to
`const char HTTP_PARAMSAVED[]       PROGMEM = "<meta http-equiv='refresh' content='5;url=/' /><div class='msg S'>Saved<br/></div>";`
basicly just a refresh after 5sec.

if you are not using a params page i guess you may need to edit this line instead?:
`const char HTTP_SAVED[]            PROGMEM = "<div class='msg'>Saving Credentials<br/>Trying to connect ESP to network.<br />If it fails reconnect to AP to try again</div>";`

tablatronix, 2023-05-23:
hmm, I guess I can add this in, I mean if saving wifi it wont do anything since the AP drops out. But do we want it to do the same for parameters? do we go back to params, or home? Do we just add a home button instead?

nospam2k, 2023-05-23:
Yea, I agree, where to go next? For me, going back to the home menu seems the most reasonable. Tasmota reboots and goes back to home for almost everything so I've gotten used to that. On an ESP thats just a few seconds.

nospam2k, 2023-05-23:
Just a thought, but it might be nice to add this into the web portal as a param on init

Halvhjearne, 2023-05-23:
i agree, if there is not already an option for this, it would be nice to have one.
idk about a home button, personally i would not really see any use for it with this.
i think most logical would be just return to main page after a short while, i feel like when you hit save, you already changed everything you needed?
obviously if wifi is saved it makes no difference what it does after, but i guess if wifi fails to connect it could also be usefull, since it would reload the main page at some point when ap is back up again.
if you want to "go wild" with it, you could scrap the whole "saved" landing page, instantly return to main page and have the saved message be like a toast instead.