tzapu/WiFiManager #1608: implicit conversion from 'float' to 'double' [-Wdouble-promotion]

fixus971, 2023-05-16:
Hardware
WiFimanager Branch/Release:  Master
Esp32

Problem description:
warning: implicit conversion from 'float' to 'double' to match other operand of binary expression [-Wdouble-promotion]

Line 2269:
    p.replace(FPSTR(T_2),(String)((temperatureRead()+32)*1.8));

*1.8 Issue at end of line

Solution
 replace with *1.8f

tablatronix, 2023-05-17:
I don't get these warnings

`build_flags: -Wall -Wdouble-promotion`

what flags?

closed 2023-05-17 by tablatronix

fixus971, 2023-05-17:
Hi. Is warning of compiler in c17 version.

even if normally this warning is not signaled.. it is still a problem for
32 bit CPUs.

I corrected many similar arithmetic calculations because I saw the
explanation and I tested it: the 32bit CPUs of the Arduino world suffer a
lot from double counts that go beyond 32bit.  Needed .. but just put an "f"
and everything falls into place

fixus971, 2023-05-17:
Some info about double calc on arduino hardware:
https://forum.arduino.cc/t/8bit-vs-32bit-floating-point-calculations/543123/11

I see that int and a float is 4 bytes and a double is 8 bytes.
after that i read and tested that double require more software time. (Double)

I used platformio with:

[env:Esp32]
framework = arduino
platform = espressif32
board = az-delivery-devkit-v4
upload_speed = 921600

monitor_speed = 115200
monitor_filters = direct  ;EscChar to console wo filtering = colors

build_unflags =
  -std=c++11
  -std=gnu++11
  -Os

build_flags   =
  -DCORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_DEBUG  ;=4
  -O3
  -std=c++17
  -std=gnu++17
  -Wall     ;en warnings questionable constructions that are easy to avoid
  -Wextra   ;en extra warning flags that are not enabled by -Wall.
  -Wdouble-promotion             ; warn double soft64b:.9(def=double)->.9f
  -Wimplicit-fallthrough         ; warn if switch case without break
  -Wreturn-type -Werror=return-type  ; no return(x) -> CPU Exception hard to found

tablatronix, 2023-05-17:
Thanks for that!

commit e9bbb6c5 (2023-05-17): fixes #1608