tzapu/WiFiManager #1609: Current WiFi Manager version does not allow to add credentials -> "No AP set" is shown only

AWSW-de, 2023-05-18:
### Basic Infos
Current WiFi Manager version does not allow to add credentials -> "No AP set" is shown only

#### Hardware
WiFimanager Branch/Release:  Master

Esp32: 
Hardware: D1 mini ESP32
Arduino 2.1.0

### Description
Current WiFi Manager version does not allow to add credentials -> "No AP set" is shown only

After setting up a new ESP32 or deleting the set WiFi credentials with "WiFi.disconnect();" ... and "manager.resetSettings();" the user is not able to set the WiFi credentials in the captive portal page. The user is shown "No AP set" is shown only.

Using an older version of the WiFi Manager library (tried 2.0.14-beta ~ October 2022 and one from ~ 2 weeks ago) fixes this problem.

### Settings in IDE
Module: Wemos D1 mini ESP32

Additional libraries: 
// # - Adafruit NeoPixel      // by Adafruit:                     https://github.com/adafruit/Adafruit_NeoPixel
// # - AsyncTCP               // by me-no-dev:                    https://github.com/me-no-dev/AsyncTCP
// # - ESPAsyncWebServer      // by me-no-dev:                    https://github.com/me-no-dev/ESPAsyncWebServer
// # - ESPUI                  // by s00500:                       https://github.com/s00500/ESPUI
// # - ArduinoJson            // by bblanchon:                    https://github.com/bblanchon/ArduinoJson
// # - LITTLEFS               // by lorol:                        https://github.com/lorol/LITTLEFS
// # - UniversalTelegramBot   // by witnessmenow:                 https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot

Thanks in advance :)

tablatronix, 2023-05-18:
Hmm html is just missing? Ruhroh I bet that last commit broke it

tablatronix, 2023-05-18:
Can you get any serial logs? And use a basic example?

AWSW-de, 2023-05-18:
Hi, 
thanks for your reply. 

Just tested with your "Basic.ino" sketch from the current download. Result is the same: "No AP set..."

Here is the serial output:

`*wm:AutoConnect 
Brownout detector was triggered
ets Jul 29 2019 12:21:46
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4
*wm:AutoConnect 
*wm:No wifi saved, skipping 
*wm:AutoConnect: FAILED for  101 ms
*wm:StartAP with SSID:  AutoConnectAP
*wm:AP IP address: 192.168.4.1
*wm:Starting Web Portal 
E (3584) wifi:AP has neither DSSS parameter nor HT Information, drop it
E (3630) wifi:AP has neither DSSS parameter nor HT Information, drop it`

Blaxxun75, 2023-05-18:
I have the same problem. I also used the "Basic.ino" and the serial shows:

`*wm:AutoConnect
*wm:No wifi saved, skipping
*wm:AutoConnect: FAILED for  23 ms
*wm:StartAP with SSID:  AutoConnectAP
*wm:AP IP address: 192.168.4.1
*wm:Starting Web Portal
[ 39789][E][WebServer.cpp:649] _handleRequest(): request handler not found
[ 39939][E][WebServer.cpp:649] _handleRequest(): request handler not found`

Html shows only "No AP Set"

Thanks!

zheins, 2023-05-18:
I'm getting this same issue using the non-blocking and basic examples, looks just like the screenshot above. I tested on a featherV2 and and edgebox-esp-100, both of which were working fine a few weeks ago on the same code.

zheins, 2023-05-18:
I'll also add that going back to commit 297123b resolved the issue for me.

Blaxxun75, 2023-05-19:
Hello, what Link do I have to give if I want a previous working version in platform.io ?
I tried a lot but I always get an error. Only the Link below seems to work, but that's the current latest version

lib_deps = https://github.com/tzapu/WiFiManager.git

Thank you

chriskinal, 2023-05-19:
According to platformio.ini docs one can specify a particular commit like so:
https://github.com/tzapu/WiFiManager/commit/88c378f0a24a99e7b9a007ea31ea43cacce924ac

tablatronix, 2023-05-19:
#1610

closed 2023-05-19 by tablatronix

Blaxxun75, 2023-05-19:
> According to platformio.ini docs one can specify a particular tag like so:
> 
> https://github.com/tzapu/WiFiManager.git#v2.0.16-rc.2

Hello, thank you very much!!
Although i read the docs too i might have used the wrong version notation

AWSW-de, 2023-05-19:
> Its already fixed

Thanks a lot!

tablatronix, 2023-05-20:
I made rc2 public, pio should use that as latest now? Let em make sure

tablatronix, 2023-05-20:
well shoot looks like it still blocks beta versions ugh

Blaxxun75, 2023-05-20:
I was able to choose a network and enter the network password.
But the ESP32 does not connect to the network...

Seirakos, 2023-05-23:
I tried using the code from basic example, I first compiled and uploaded through arduino IDE, it worked fine, but then I tried to do the same through platformio, and I keep getting this error.
`*wm:resetSettings 
*wm:SETTINGS ERASED 
*wm:AutoConnect
*wm:No wifi saved, skipping 
*wm:AutoConnect: FAILED for  22 ms
*wm:StartAP with SSID:  AutoConnectAP
*wm:AP IP address: 192.168.4.1
*wm:Starting Web Portal
[ 23537][E][WebServer.cpp:649] _handleRequest(): request handler not found`

tablatronix, 2023-05-23:
*wm:resetSettings  *wm:SETTINGS ERASED ??

Seirakos, 2023-05-24:
Yes, I tried it without that uncommented and it was pretty much the same.
Is there another dependency needed for using WifiManager on esp32 when working with PlatformIO? Arduino IDE works just fine but I can't stand the compile times there.

tablatronix, 2023-05-24:
I mean thats not an error thats exactly what it is supposed to do, its says starting web portal, whats the problem?