tzapu/WiFiManager #1610: Root menu always empty after commit b735c61

nonsintetic, 2023-05-18:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32: 

Hardware:			ESP32

### Description

The menu in the root page is always empty. Think I figured out why:

Commit b735c61 moved the menu tokens to PROGMEM, changing variable types slightly.  This caused the comparison done in `WiFiManager::setMenu()` to determine what menus to always equals false now. This is because it's now comparing pointers, memory addresses, instead of contents like before. 

See comment in code below:
```
void WiFiManager::setMenu(const char * menu[], uint8_t size){
  _menuIds.clear();
  for(size_t i = 0; i < size; i++){
    for(size_t j = 0; j < _nummenutokens; j++){
      //following line compares memory addresses now, not contents like before
      if(menu[i] == _menutokens[j]){
        if((String)menu[i] == "param") _paramsInWifi = false; // param auto flag
        _menuIds.push_back(j);
      }
    }
  }
}

```

tablatronix, 2023-05-19:
Yeah looks like I changed the consts def but not the methods that use it crap

tablatronix, 2023-05-19:
Also need to check that nummenutokens size is set

tablatronix, 2023-05-19:
9b8de0d

closed 2023-05-19 by tablatronix