tzapu/WiFiManager #1614: Erase Configuration, setShowStaticFields and dynamic/ static Ip setting

kutty1999, 2023-05-24:
WiFimanager Branch/Release:       2.0.16-rc.2

Esp8266/Esp32:        Esp8266

Hardware:        ESP-12e

Core Version:       3.1.2

### Description
I have no idea from get dns from the field tab in webpage and store in json in code i declared default dns as 8.8.8.8 
then I need same as above for secondary dns.

I have faced three problems.
The first one, on demand, I tried to erase the older configuration but it didn't work.
The second one is static,dns, and custom parameter fields disappear after the initial configuration I declared the true value in the function. it will reappear after erase the configuration from the webpage.
The third one is not switching between dynamic and static IP iam trying to switch the IP mode but it does not make it.

### Settings in IDE

Module:  			NodeMcu

Additional libraries: PubSubClient, ESP8266WiFi

### Sketch

```cpp
#include <FS.h>                   //this needs to be first, or it all crashes and burns...
#include <ESP8266WiFi.h>
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager
#include <PubSubClient.h>       // PubSub Library used for MQTT protocol must needed very important.
#include <ESP8266WebServer.h>
#include <DNSServer.h>
#include <ArduinoJson.h>          //https://github.com/bblanchon/ArduinoJson
#include <ESPping.h>
#ifdef ESP32
  #include <SPIFFS.h>
#endif

#define buttonPin 12    // Reset button pin
#define LONG_PRESS_TIME 5000   // long press of reset button time in milliseconds 5000
const char* remote_host = "www.google.com"; //Host name to ping for internet connectivity.
char mqtt_server[40] = "xxxx";
char user[20] = "xxxx";             // Username of MQTT Broker
char pass[30] = "xxxxx";           // Password of MQTT Broker
int wifiStatus;                   // TO check the WiFi status.
WiFiClient espClient;                    
PubSubClient client(espClient);

WiFiManager wifiManager;

//default custom static IP
char static_ip[16]="10.0.1.56";
char static_gw[16]="10.0.1.1";
char static_sn[16]="255.255.255.0";
char dns[16]="8.8.8.8";
//flag for saving data
bool shouldSaveConfig = false;

//callback notifying us of the need to save config
void saveConfigCallback () {
  Serial.println("Should save config");
  shouldSaveConfig = true;
}
 
void reconnect() {
  // Loop until we're reconnected
 if (!client.connected()) { 
    Serial.print("Attempting MQTT connection...");
    digitalWrite(BUILTIN_LED, LOW);
    // Create a random client ID
[... 302 lines cut ...]
Subnet Mask: 255.255.255.0
*wm:AP IP address: 192.168.4.1
*wm:Starting Web Portal 
*wm:9 networks found
*wm:STA IP set: 10.0.1.56
*wm:Connecting to NEW AP: Milanity Production
*wm:connectTimeout not set, ESP waitForConnectResult... 
*wm:Connect to new AP [SUCCESS] 
*wm:Got IP Address: 
*wm:10.0.1.56 
Should save config
*wm:config portal exiting 
connected...yeey :)
saving config
{"mqtt_server":"milancloud.com","ip":"10.0.1.56","gateway":"10.0.1.1","subnet":"255.255.255.0"}

Connected to xxxxx
Local IP: 10.0.1.56
Gateway IP: 10.0.1.1
Subnet Mask: 255.255.255.0
Attempting MQTT connection...failed, rc=-2 try again in 1 second
Attempting MQTT connection...failed, rc=-2 try again in 1 second
Attempting MQTT connection...failed, rc=-2 try again in 1 second
Attempting MQTT connection...failed, rc=-2 try again in 1 second
```

kutty1999, 2023-05-26:
> WiFimanager Branch/Release: 2.0.16-rc.2
> 
> Esp8266/Esp32: Esp8266
> 
> Hardware: ESP-12e
> 
> Core Version: 3.1.2
> 
> ### Description
> I have no idea from get dns from the field tab in webpage and store in json in code i declared default dns as 8.8.8.8 then I need same as above for secondary dns.
> 
> I have faced three problems. The first one, on demand, I tried to erase the older configuration but it didn't work. The second one is static,dns, and custom parameter fields disappear after the initial configuration I declared the true value in the function. it will reappear after erase the configuration from the webpage. The third one is not switching between dynamic and static IP iam trying to switch the IP mode but it does not make it.
> 
> ### Settings in IDE
> Module: NodeMcu
> 
> Additional libraries: PubSubClient, ESP8266WiFi
> 
> ### Sketch
> ```c++
> #include <FS.h>                   //this needs to be first, or it all crashes and burns...
> #include <ESP8266WiFi.h>
> #include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager
> #include <PubSubClient.h>       // PubSub Library used for MQTT protocol must needed very important.
> #include <ESP8266WebServer.h>
> #include <DNSServer.h>
> #include <ArduinoJson.h>          //https://github.com/bblanchon/ArduinoJson
> #include <ESPping.h>
> #ifdef ESP32
>   #include <SPIFFS.h>
> #endif
> 
> #define buttonPin 12    // Reset button pin
> #define LONG_PRESS_TIME 5000   // long press of reset button time in milliseconds 5000
> const char* remote_host = "www.google.com"; //Host name to ping for internet connectivity.
> char mqtt_server[40] = "xxxx";
> char user[20] = "xxxx";             // Username of MQTT Broker
> char pass[30] = "xxxxx";           // Password of MQTT Broker
> int wifiStatus;                   // TO check the WiFi status.
> WiFiClient espClient;                    
> PubSubClient client(espClient);
> 
> WiFiManager wifiManager;
> 
> //default custom static IP
> char static_ip[16]="10.0.1.56";
> char static_gw[16]="10.0.1.1";
> char static_sn[16]="255.255.255.0";
> char dns[16]="8.8.8.8";
> //flag for saving data
> bool shouldSaveConfig = false;
> 
> //callback notifying us of the need to save config
> void saveConfigCallback () {
>   Serial.println("Should save config");
>   shouldSaveConfig = true;
> }
>  
> void reconnect() {
>   // Loop until we're reconnected
>  if (!client.connected()) { 
>     Serial.print("Attempting MQTT connection...");
>     digitalWrite(BUILTIN_LED, LOW);
>     // Create a random client ID
>     String clientId = "ESP8266Client-";   
>     clientId += String(random(0xffff), HEX);   /*************************** Client Id generated ramdomly *************************/
>     // Attempt to connect
>     if (client.connect(clientId.c_str(),user,pass)) { /*********************** Connection init to MQTT server *********/
>        client.subscribe(sub_topic);  /********************************** Subscribe starts here *****************************/
>        wifiStatus = WiFi.status();  
[... 295 lines cut ...]
> Subnet Mask: 255.255.255.0
> *wm:AP IP address: 192.168.4.1
> *wm:Starting Web Portal 
> *wm:9 networks found
> *wm:STA IP set: 10.0.1.56
> *wm:Connecting to NEW AP: xxx
> *wm:connectTimeout not set, ESP waitForConnectResult... 
> *wm:Connect to new AP [SUCCESS] 
> *wm:Got IP Address: 
> *wm:10.0.1.56 
> Should save config
> *wm:config portal exiting 
> connected...yeey :)
> saving config
> {"mqtt_server":"milancloud.com","ip":"10.0.1.56","gateway":"10.0.1.1","subnet":"255.255.255.0"}
> 
> Connected to xxxxx
> Local IP: 10.0.1.56
> Gateway IP: 10.0.1.1
> Subnet Mask: 255.255.255.0
> Attempting MQTT connection...failed, rc=-2 try again in 1 second
> Attempting MQTT connection...failed, rc=-2 try again in 1 second
> Attempting MQTT connection...failed, rc=-2 try again in 1 second
> Attempting MQTT connection...failed, rc=-2 try again in 1 second
> ```

tablatronix, 2023-05-31:
I have no idea what this is about?

kutty1999, 2023-05-31:
> I have no idea what this is about?

Thanks for reply @tablatronix 
I try to create a AutoConnect With FSParameters and CustomIP with ondemand to Starting Web Portal for configuration but i couldn't achieve my idea 

I tried to erase the older configuration before starting web portal but it didn't work.

```cpp
void loop() {
  static unsigned long buttonPressStartTime = 0;
  static bool buttonPressed = false;

  if (digitalRead(buttonPin) == HIGH) {
    if (!buttonPressed) {
      Serial.println("AP Button is Pressed");
      buttonPressed = true;
      buttonPressStartTime = millis();
    }
    if (millis() - buttonPressStartTime > LONG_PRESS_TIME) {
      WiFi.disconnect();
      WiFi.mode(WIFI_AP);
      WiFi.softAP("AP Mode");
      WiFiManager wifiManager;
      wifiManager.resetSettings();      //reset settings  
      wifiManager.startConfigPortal("AP Mode");
    }
      }
   else {
    buttonPressed = false;
  }
  ```
static,dns, and custom parameter fields disappear after the first configuration. I declared the true boolean value in the function. it will reappear after erase the configuration using erase config button in the webpage.

```cpp
  wifiManager.setShowStaticFields(true);  // show Static IP  user Field 
  wifiManager.setShowDnsFields(true); // show DNS  user Field 
```
Not switching between dynamic and static IP iam trying to switch the IP mode but it does not make it. 
I conform the dynamic not setting by the I unfill the static, and dns Fields it goes to custom ip configuration It take ip,gateway, subnet mask from the json file of older configuration, once set the static after unable to switch dynamic. 

```cpp
if (json["ip"]) {
            Serial.println("setting custom ip from config");
            strcpy(static_ip, json["ip"]);
            strcpy(static_gw, json["gateway"]);
            strcpy(static_sn, json["subnet"]);
            Serial.println(static_ip);
          } else {
            Serial.println("no custom ip in config");
          }
        } else {
          Serial.println("failed to load json config");
```

tablatronix, 2023-05-31:
Someone else did this, I think they added a checkbox.

Are you adding them to wm? wifiManager.setSTAStaticIPConfig(_ip, _gw, _sn);

did you check the examples?