tzapu/WiFiManager #1615: [ERROR] WiFiManagerParameter is out of scope when using nonblocking

mazWaz, 2023-05-24:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32: 

Hardware:			ESP32-Devkit

Core Version:      	2.4.0, staging

### Description

Problem description
i using examples/Super/OnDemandConfigPortal/OnDemandConfigPortal.ino
bool TEST_CP = false;
bool WMISBLOCKING = false; 
but addParameter not show on browser and error like this 
```wm:[1] [ERROR] WiFiManagerParameter is out of scope```
and my custom debug
```
wm:[4] Param _length 1073541520
*wm:[4] Param _label
*wm:[4] Param _labelPlacement
```

### Settings in IDE

Module:  			ESP32-Devkit

Additional libraries: 

### Sketch custom debug on String WiFiManager::getParamOut()

```cpp
#BEGIN
for (int i = 0; i < _paramsCount; i++)
    {
      // Serial.println((String)_params[i]->_length);
      DEBUG_WM(DEBUG_DEV, F("Param _length"), _params[i]->_length);
      DEBUG_WM(DEBUG_DEV, F("Param _label"), _params[i]->_label);
      DEBUG_WM(DEBUG_DEV, F("Param _labelPlacement"), _params[i]->_labelPlacement);
      if (_params[i] == NULL || _params[i]->_length > 99999)
      {
// try to detect param scope issues, doesnt always catch but works ok
#ifdef WM_DEBUG_LEVEL
        DEBUG_WM(DEBUG_ERROR, F("[ERROR] WiFiManagerParameter is out of scope"));
#endif
        return "";
      }
    }
#END
```

### Debug Messages

```
wm:[1] [ERROR] WiFiManagerParameter is out of scope
wm:[4] Param _length 1073541520
*wm:[4] Param _label
*wm:[4] Param _labelPlacement
```

BlaineAtkins, 2023-05-24:
I believe your issue is that you're declaring the things like
` WiFiManagerParameter custom_mqtt_server("server", "mqtt server", "", 40);`
in setup() instead of in the global scope. See if that solves the issue.

If you want an example, here's a project I made that successfully combines custom parameters with a non-blocking portal
https://github.com/BlaineAtkins/GenericOutletBox/blob/main/generic_wifi_box.ino

mazWaz, 2023-05-25:
> 

> I believe your issue is that you're declaring the things like ` WiFiManagerParameter custom_mqtt_server("server", "mqtt server", "", 40);` in setup() instead of in the global scope. See if that [...]

### Solved
need define to global scope or create new global variable

closed 2023-05-25 by mazWaz