tzapu/WiFiManager #1618: How to disable blinking LED? I need to use the GPIO pin.

rtek1000, 2023-06-01:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32: 

Hardware:			ESP32

Core Version:      	2.0.8  (https://github.com/espressif/arduino-esp32)

### Description

Problem description

- LED continues to flash even after station mode has been set.

- But there is no LED declaration in the source code.

- If the setup line "res = wm.autoConnect("AutoConnectAP", "password");" is commented out, the LED stops flashing.

- I found a mention of LED in the issues:

> Callback during wifi setup loop (for example, to flash an LED)

- https://github.com/tzapu/WiFiManager/pull/522

### Settings in IDE

Module:  			ESP32 WROOM DA

Additional libraries: 

### Sketch

```cpp
#BEGIN
#include <Arduino.h>

void wifiManager_setup(void) {
  pinMode(WM_TRIGGER_PIN, INPUT);

  // wm.resetSettings(); // wipe settings

  if (wm_nonblocking) wm.setConfigPortalBlocking(false);

  // add a custom input field
  int customFieldLength = 40;

  // new (&custom_field) WiFiManagerParameter("customfieldid", "Custom Field Label", "Custom Field Value", customFieldLength,"placeholder=\"Custom Field Placeholder\"");

  // test custom html input type(checkbox)
  // new (&custom_field) WiFiManagerParameter("customfieldid", "Custom Field Label", "Custom Field Value", customFieldLength,"placeholder=\"Custom Field Placeholder\" type=\"checkbox\""); // custom html type

  // test custom html(radio)
  const char* custom_radio_str = "<br/><label for='customfieldid'>Custom Field Label</label><input type='radio' name='customfieldid' value='1' checked> One<br><input type='radio' name='customfieldid' value='2'> Two<br><input type='radio' name='customfieldid' value='3'> Three";
  new (&custom_field) WiFiManagerParameter(custom_radio_str); // custom html input

  wm.addParameter(&custom_field);
  wm.setSaveParamsCallback(saveParamCallback);

  // custom menu via array or vector
  //
  // menu tokens, "wifi","wifinoscan","info","param","close","sep","erase","restart","exit" (sep is seperator) (if param is in menu, params will not show up in wifi page!)
  // const char* menu[] = {"wifi","info","param","sep","restart","exit"};
  // wm.setMenu(menu,6);
  std::vector<const char *> menu = {"wifi", "info", "param", "sep", "restart", "exit"};
  wm.setMenu(menu);

  // set dark theme
  wm.setClass("invert");
  
  //set static ip
  // wm.setSTAStaticIPConfig(IPAddress(10,0,1,99), IPAddress(10,0,1,1), IPAddress(255,255,255,0)); // set static ip,gw,sn
  // wm.setShowStaticFields(true); // force show static ip fields
  // wm.setShowDnsFields(true);    // force show dns field always

  // wm.setConnectTimeout(20); // how long to try to connect for before continuing
  wm.setConfigPortalTimeout(30); // auto close configportal after n seconds
  // wm.setCaptivePortalEnable(false); // disable captive portal redirection
  // wm.setAPClientCheck(true); // avoid timeout if client connected to softap

  // wifi scan settings
  // wm.setRemoveDuplicateAPs(false); // do not remove duplicate ap names (true)
  // wm.setMinimumSignalQuality(20);  // set min RSSI (percentage) to show in scans, null = 8%
  // wm.setShowInfoErase(false);      // do not show erase button on info page
  // wm.setScanDispPerc(true);       // show RSSI as percentage not graph icons

  // wm.setBreakAfterConfig(true);   // always exit configportal even if wifi save fails

  bool res;
  // res = wm.autoConnect(); // auto generated AP name from chipid
  // res = wm.autoConnect("AutoConnectAP"); // anonymous ap
  res = wm.autoConnect("AutoConnectAP", "password"); // password protected ap // <------------------------- this line

  if (!res) {
    Serial.println("Failed to connect or hit timeout");
    // ESP.restart();
  } else {
    //if you get here you have connected to the WiFi
    Serial.println("connected...yeey :)");
  }
}

void loop() {

}
#END
```

### Debug Messages

```
messages here
```

tablatronix, 2023-06-01:
What board?

rtek1000, 2023-06-01:
> What board? What led?

ESP32-WROOM-DA, GPIO2.

[image]

chriskinal, 2023-06-01:
I had trouble with board profiles that used the blue LED causing WDT timeouts. While troubleshooting I changed Arduino IDE to the ESP32 Dev Module profile. This stopped the use of the blue LED and the WDT timeouts. I could still use the GPIO2 to activate the blue LED.

tablatronix, 2023-06-01:
you are using the wrong board defs for this board, thats the problem

closed 2023-06-01 by tablatronix

rtek1000, 2023-06-01:
I was testing it too, and I noticed that if other board models (i.e. DOIT ESP32 DEV KIT V1; ESP32 DEV Module) are selected in the IDE, the LED (GPIO2) stops blinking, so it seems to be only with the ESP32-WROOM-DA. I'll leave the reference of this model, in case someone looks for it:

[image]

https://www.techtonions.com/esp32-wroom-da/

rtek1000, 2023-06-01:
>The ESP32-WROOM-DA module uses GPIO 2 and GPIO 25 to select which antenna is active. These are referenced from the datasheet for the module in table 3 (just after the pin assignment list).

Source: https://github.com/espressif/arduino-esp32/issues/8280

tablatronix, 2023-06-01:
I have never seen any DA board than the esp devkit, what board exactly is this?

tablatronix, 2023-06-01:
I knew IT

```
The ESP32-WROOM-DA module uses GPIO 2 and GPIO 25 to select which antenna is active. These are referenced from the datasheet for the module in table 3 (just after the pin assignment list). Make sure you are selecting “ESP32-WROOM-DA Module” in the boards menu which has additional defines added to enable dual antenna mode support.
```

why the fuck was this so hard to find in the docs..

rtek1000, 2023-06-01:
The price for having a more powerful signal was losing 2 general purpose pins.

Since the GPIO 25 is DAC1.

D'oh!

¯\_(ツ)_/¯

tablatronix, 2023-06-01:
SO how is the signal ?

I wil have to order one of these to test

rtek1000, 2023-06-01:
I found this video about the performance,

See more about this ESP32-Shirt:

#422 A new Dual Antenna ESP32 Module: The WROOM-DA
https://m.youtube.com/watch?v=F0u5qIwwY1k&vl=en

tablatronix, 2023-06-03:
So you haven't used it? I thought you said thats what you are testing?