tzapu/WiFiManager #1620: Other processes stops until the device is paired with the Wifi

patelabhi23, 2023-06-05:
My microcontroller has Wifi Manager integrated in it & there is a sensor also connected to it. So as soon as I turn on my device & until the device is configured with Wifi networks available. The microcontroller isn't being able to record the data from the sensor. Only after device is configured with wifi then only sensor readings are being done. 

## Instead of above what it should be:

The microcontroller should be able to record the data of sensors & along with that parallely wifi manager should be running to.

@tablatronix is there any solution for this?

tablatronix, 2023-06-06:
Who code is it? 

https://github.com/tzapu/WiFiManager/wiki/Methods#setconfigportaltimeoutunsigned-long-seconds-green_book

xr1140, 2023-08-23:
I think you should run WiFiManager in non-blocking mode. using 
`wm.setConfigPortalBlocking(false);`

closed 2023-08-24 by tablatronix