tzapu/WiFiManager #1622: Force MDNS restart after connection (NONBlockling)

Shumatic, 2023-06-07:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master 2.0.16-rc.2

Esp8266: 

Hardware:			ESP-12e

Core Version:      	2.4.0, staging

### Description

Is there a callback function to indicate that a connection is made?
When using NONblocking and MDNS, it connects fine, but MDNS takes a long time to initialize,
so i would like to force a MDNS restart.

### Settings in IDE 1.8.13

Module:  			Generic

Shumatic, 2023-06-09:
> hmm on esp8266 you have to explicitly run mdns, maybe there is a problem there.

No, i am aware of this, and i have MDNS.begin ( "name" ) ) in setup.
If ESP dosnt have a WiFi connection when booted, it still setup MDNS, and when i configure WiFi through the WiFimaneger it connects, but it isn't possile to reach from "name" only ip. It takes more then an hour before it works.

tablatronix, 2023-06-09:
If you use WM non blocking it handles this for you.
Otherwise I think you are suppose to call this in loop

'MDNS.update();'

Maybe this is outdated, I will have to check the docs and code for esp mdns.

But give that a try

Shumatic, 2023-06-10:
> 
Thanks
When i reboot after i have set WiFi, it works fine.
I have tried to force both MDNS.update() and MDNS.begin(xxx) with a botton, but it dosnt work.
Shouldnt the ESP restart automatically after WiFi have benn set? 
i think that would solve it.

tablatronix, 2023-06-10:
mdns stop start?

Shumatic, 2023-06-11:
Hi Thers no Start stop mdns....

tablatronix, 2023-06-11:
.end() sorry

Shumatic, 2023-06-14:
I havent tested properly...
when i set WiFi throug WiFimanager, and it reboots, i cant connect to it with IP either, i need to restart. so it isnt a MDNS problem

tablatronix, 2023-06-14:
Thats strange, esp lib up to date?

Shumatic, 2023-07-04:
> Thats strange, esp lib up to date?

Thanks for still helping me out.....
Hi, tested some more, and it is the same, when i put in credentials, it connect and get NTPsync, but i cant connect, even throug IP .
Another thing, if I put in a wrong password, and connect fails, it reboot portal, and i can correct PW, but it still dosnt work, i have to go in portal and ERASE WiFIconfig, then i can set theright PW.