tzapu/WiFiManager #1623: Customfields saved values not display on portal

zstergios, 2023-06-08:
When I configure the settings are saved and everything works great, if I start the portal to change some settings, the last saved values are not display and default value is displayed.

The thrid parameter is default value. 
I expected to see the last saved value instead. Is it bug or I missing something?

```cpp
#BEGIN
  WiFiManagerParameter custom_weburl("weburl", "WebPOS URL","https://NAME.domain.com", 128, "placeholder=\"Web URL\"");
  wm.addParameter(&custom_weburl);

  wm.setSaveParamsCallback(saveParamCallback);
  wm.setMinimumSignalQuality(10);
  wm.setConfigPortalTimeout(180);
  //wm.setConfigPortalBlocking(true);
#END
```

### Debug Messages

```
messages here
```

tablatronix, 2023-06-09:
you have to restore saved values, wm doesnt know what you saved

zstergios, 2023-06-09:
**How to restore them?**

This way causes restarts

`String getParam(String name){
  //read parameter from server, for customhmtl input
  String value;
  if(wm.server->hasArg(name)) {
    value = wm.server->arg(name);
  }
  return value;
}`

and that way returns empty value probably because it's before adding the "custom_weburl" to  wm.addParameter

  const char* url =  custom_weburl.getValue();
  WiFiManagerParameter custom_weburl("weburl", "WebPOS URL",url, 128, "placeholder=\"Web URL\"");

zenz, 2024-03-04:
>   if(wm.server->hasArg(name)) {
>     value = wm.server->arg(name);
>   }
>   return value;
> }
there's a bug, since wm.server at that moment is equal to nullptr. then you cannot get the custom_field value.

tablatronix, 2024-03-05:
What do you mean? Thats called in a callback whats the scope here?

zenz, 2024-03-05:
> What do you mean? Thats called in a callback whats the scope here?

you can just try to compile and run the Advanced samples, while using custom_field, if you want to get custom_field value, it will crash, since wm.server is equal to nullptr.

tablatronix, 2024-03-05:
Still not sure what you are calling, this happens in the example with no changes?

zenz, 2024-03-06:
while compile and running this sample
https://github.com/tzapu/WiFiManager/blob/master/examples/Advanced/Advanced.ino

you will find that custom field value  with name "customfieldid" cannot be get and store, at that moment calling wm.server, it's nullptr. you will never have a change to use hasArg or arg to check and get the value.

tablatronix, 2024-03-08:
But its only called from the callback is it not?