tzapu/WiFiManager #1626: Error compiling for board WiFi Kit 32(V3)

PKeyes505, 2023-06-16:
I am fairly new and any help would be appreciated. 

I am using the Heltec WiFi Kit 32(V3) with the Arduino IDE version 1.8.19 and I am able to load other sketches that utilize wifi, OTA, Etc..  I have tried to eliminate many variables as possible with a fresh virtual development environment with just the WiFi Manager library and HelTec Board installed but continue to get the "Error compiling for board WiFi Kit 32(V3)." Error.

```c:/users/pdkey/appdata/local/arduino15/packages/heltec-esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\pdkey\AppData\Local\Temp\arduino_cache_213280\core\core_f796d41ba73a20c2b91f9a376b5bb492.a(esp32-hal-misc.c.o):(.literal.temperatureRead+0x4): undefined reference to `temp_sensor_set_config'
c:/users/pdkey/appdata/local/arduino15/packages/heltec-esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\pdkey\AppData\Local\Temp\arduino_cache_213280\core\core_f796d41ba73a20c2b91f9a376b5bb492.a(esp32-hal-misc.c.o):(.literal.temperatureRead+0x8): undefined reference to `temp_sensor_start'
c:/users/pdkey/appdata/local/arduino15/packages/heltec-esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\pdkey\AppData\Local\Temp\arduino_cache_213280\core\core_f796d41ba73a20c2b91f9a376b5bb492.a(esp32-hal-misc.c.o):(.literal.temperatureRead+0xc): undefined reference to `temp_sensor_read_celsius'
c:/users/pdkey/appdata/local/arduino15/packages/heltec-esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\pdkey\AppData\Local\Temp\arduino_cache_213280\core\core_f796d41ba73a20c2b91f9a376b5bb492.a(esp32-hal-misc.c.o):(.literal.temperatureRead+0x10): undefined reference to `temp_sensor_stop'
c:/users/pdkey/appdata/local/arduino15/packages/heltec-esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\pdkey\AppData\Local\Temp\arduino_cache_213280\core\core_f796d41ba73a20c2b91f9a376b5bb492.a(esp32-hal-misc.c.o): in function `temperatureRead':
C:\Users\pdkey\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\cores\esp32/esp32-hal-misc.c:71: undefined reference to `temp_sensor_set_config'
c:/users/pdkey/appdata/local/arduino15/packages/heltec-esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\pdkey\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\cores\esp32/esp32-hal-misc.c:72: undefined reference to `temp_sensor_start'
c:/users/pdkey/appdata/local/arduino15/packages/heltec-esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\pdkey\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\cores\esp32/esp32-hal-misc.c:73: undefined reference to `temp_sensor_read_celsius'
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\pdkey\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
c:/users/pdkey/appdata/local/arduino15/packages/heltec-esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\pdkey\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\cores\esp32/esp32-hal-misc.c:74: undefined reference to `temp_sensor_stop'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board WiFi Kit 32(V3).
```

tablatronix, 2023-06-17:
Looks like something is out of date, update arduino and libraries

PKeyes505, 2023-06-17:
I am running Arduino 1.8.19 and all installed libraries have been updated and are now at the latest version.  I am trying to verify the Basic.ino example and am still getting the Error compiling for board WiFi Kit 32(V3).  Other suggestions?

chriskinal, 2023-06-17:
Looks like the development environment is missing a library for some sort of temperature sensor. Not sure why the WiFI manager example would trigger the issue and not other sketches but the errors point to an issue with the Heltec packages.

Try clearing the cache when you update libraries or packages since the IDE caches stuff to speed up compilation. Google for the process to clear the Arduino IDE cache since it is not straight forward. The cache issue drove me nuts trying to troubleshoot an issue. I would update libraries and still have the issue. Turned out the IDE was not seeing the library was updated so it did not recompile using the new library. It kept using the cached version.

tablatronix, 2023-06-18:
I saw local/arduino15/ and was wondering...

Yeah using dev in arduino is problematic, why I switched to pio

chriskinal, 2023-06-18:
That and cleaning the cache is a pain in the Arduino IDE. I switched to VSCode/platformIO. Get integrated GIT to boot.

Mariano74-ARG, 2023-08-30:
Hi. I'm kind of newbie on arduino. Just trying wifimanager on a Heltec Lora 32 (V.3)   (ESP32S3), using Arduino IDE 2.1.1 and I'm getting several compiling errors:

c:/users/maria/appdata/local/arduino15/packages/heltec-esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\maria\AppData\Local\Temp\arduino\cores\47bf7ccab0b1915fef044f26ba8939d8\core.a(esp32-hal-misc.c.o):(.literal.temperatureRead+0x4): undefined reference to `temp_sensor_set_config'
c:/users/maria/appdata/local/arduino15/packages/heltec-esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\maria\AppData\Local\Temp\arduino\cores\47bf7ccab0b1915fef044f26ba8939d8\core.a(esp32-hal-misc.c.o):(.literal.temperatureRead+0x8): undefined reference to `temp_sensor_start'
c:/users/maria/appdata/local/arduino15/packages/heltec-esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\maria\AppData\Local\Temp\arduino\cores\47bf7ccab0b1915fef044f26ba8939d8\core.a(esp32-hal-misc.c.o):(.literal.temperatureRead+0xc): undefined reference to `temp_sensor_read_celsius'
c:/users/maria/appdata/local/arduino15/packages/heltec-esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\maria\AppData\Local\Temp\arduino\cores\47bf7ccab0b1915fef044f26ba8939d8\core.a(esp32-hal-misc.c.o):(.literal.temperatureRead+0x10): undefined reference to `temp_sensor_stop'
c:/users/maria/appdata/local/arduino15/packages/heltec-esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\maria\AppData\Local\Temp\arduino\cores\47bf7ccab0b1915fef044f26ba8939d8\core.a(esp32-hal-misc.c.o): in function `temperatureRead':
C:\Users\maria\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\cores\esp32/esp32-hal-misc.c:71: undefined reference to `temp_sensor_set_config'
c:/users/maria/appdata/local/arduino15/packages/heltec-esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\maria\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\cores\esp32/esp32-hal-misc.c:72: undefined reference to `temp_sensor_start'
c:/users/maria/appdata/local/arduino15/packages/heltec-esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\maria\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\cores\esp32/esp32-hal-misc.c:73: undefined reference to `temp_sensor_read_celsius'
c:/users/maria/appdata/local/arduino15/packages/heltec-esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: C:\Users\maria\AppData\Local\Arduino15\packages\Heltec-esp32\hardware\esp32\0.0.7\cores\esp32/esp32-hal-misc.c:74: undefined reference to `temp_sensor_stop'
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

Could you guide me to solve these issues (sensor errors and so) ?. Thank you in advance.

tablatronix, 2023-09-09:
"arduino15" ?

Mariano74-ARG, 2023-09-10:
Arduino IDE 2
The board is the Heltec Lora 32 V.3  (ESP32 S3).
I got a solution consisting on commentig the lines in the .h file included within Heltec libraries, concerning the error messages about MCPU sensors.

tablatronix, 2023-09-11:
That looks like old libraries

xznhj8129, 2024-04-27:
> That looks old or is it just an old pathname? arduino is at Arduino IDE 2.2.1
> 
> Ask in aruduino, you have mismatched esp libraries, the s2/s3 do not have temperature sensors before the latest IDF

https://forum.arduino.cc/t/what-is-arduino15-directory/576569/2?u=heckingarduinerino1