tzapu/WiFiManager #1627: How to save/load a static IP field ?

julian-lp, 2023-06-17:
I've been trying to connect to a wi fi router assigning through code the ip for the esp32 (the device will be used somewhere and I dont have access to its serial port to know IP assigned to it)

The configuration portal has one field whose name is Static IP, but I'm unclear how can I get that field and use in the line
"wm.setSTAStaticIPConfig(IPAddress(CONF_IP_LOCAL), IPAddress(CONF_IP_DEFAULT_GATEWAY), IPAddress(CONF_IP_SUBNET));"

cause it is clear that it's me who is setting 192,168,1,177   IP rather the one I entered by the config portal

Sorry for my english, hope I explained the issue almost well 

```cpp

#define CONF_IP_LOCAL 192,168,1,177    
#define CONF_IP_DEFAULT_GATEWAY 192,168,1,1  
#define CONF_IP_SUBNET 255,255,255,0 
#define CONF_SSID_INICIAL "ESP32"
#define CONF_PASS_INICIAL "ABC123456"

//in the following line I want to use the ip set in the config portal
wm.setSTAStaticIPConfig(IPAddress(CONF_IP_LOCAL), IPAddress(CONF_IP_DEFAULT_GATEWAY), IPAddress(CONF_IP_SUBNET));

    res = wm.autoConnect(CONF_SSID_INICIAL,CONF_PASS_INICIAL); // password protected ap

    if(!res){
        Serial.println("Failed to connect");
        //ESP.restart();
    }else{
        //if you get here you have connected to the WiFi    
        Serial.println("CONNECTED :)");
    }

```

julian-lp, 2023-06-19:
Well, I keep investigating a little bit and I found this:

If I include this line

wm.setSTAStaticIPConfig(IPAddress(CONF_IP_LOCAL), IPAddress(CONF_IP_DEFAULT_GATEWAY), IPAddress(CONF_IP_SUBNET));

then those fields (static IP, gateway and subnet) appears in the config portal. Otherwise, if the line isn't included, those fields arent visible in the config portal.

Now... how can I get the values of those fields ???(in order to save them to flash and retrieve every time the esp boots), given the fact that they are not "additional parameters" such as

WiFiManagerParameter parameter("parameterId", "Label", "default value", 40);

tablatronix, 2023-09-29:
I think you have to save and load static yourself