tzapu/WiFiManager #1628: Compilation Error - 'esp_wifi_set_country_code' was not declared in this scope

wagenbm, 2023-06-18:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp32: 

Hardware: ESP32-E Firebeetle

### Description

Problem description

I am not able to compile my program when using the Wifimanager. I am always getting the following error:

c:\ESP32\libraries\WiFiManager\WiFiManager.cpp: In member function 'bool WiFiManager::WiFiSetCountry()':
c:\ESP32\libraries\WiFiManager\WiFiManager.cpp:3552:11: error: 'esp_wifi_set_country_code' was not declared in this scope
     err = esp_wifi_set_country_code(_wificountry.c_str(), true);
           ^~~~~~~~~~~~~~~~~~~~~~~~~
c:\ESP32\libraries\WiFiManager\WiFiManager.cpp:3552:11: note: suggested alternative: 'esp_wifi_set_country'
     err = esp_wifi_set_country_code(_wificountry.c_str(), true);
           ^~~~~~~~~~~~~~~~~~~~~~~~~
           esp_wifi_set_country

exit status 1

I tried with downgrading the version and found out, that the problem happens as of version 2.0.12beta.

### Sketch

```
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

void setup() {
    WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP
    // it is a good practice to make sure your code sets wifi mode how you want it.

    // put your setup code here, to run once:
    Serial.begin(115200);
    
    //WiFiManager, Local intialization. Once its business is done, there is no need to keep it around
    WiFiManager wm;

    bool res;

    res = wm.autoConnect("AutoConnectAP","password"); // password protected ap

    if(!res) {
        Serial.println("Failed to connect");
        // ESP.restart();
    } 

}

void loop() {

}
```

Sukeczech, 2024-11-06:
the problem still persists even on the latest version, how to solve it?

tablatronix, 2024-11-08:
I assume the wrong libraries are being included, or its pinned to an older version of the SDK, see if there is an updated board def for that board, or remove and reinstall it, if its outdated then you might have to just use another board def

Sukeczech, 2024-11-08:
When I put the older version it works. But from 2.0.12beta up id did not.