tzapu/WiFiManager #1631: How can I erase my all saved wifi credentials from esp32

Sudheer474, 2023-06-28:
I want to reset all wifi credentials which I've configured first time but when I'm running wifiManeger.resetSettings().. it is not working because it automatically connects to the previously configured Access point. But i want is when i run  wifiManeger.resetSettings() this it then esp32 will ask me to re-configure wifi.

Hope you get my problem. 
Thanking you.

tablatronix, 2023-06-28:
hmm, whats the code look like, it should work. Maybe try setting a wifi mode first?

akutruff, 2023-10-11:
I think there's a bug in the reset code.  The struct isn't zeroed out before it's being written to memory so garbage can end up stored that appears to have an ssid.

Grey-Lancaster, 2023-10-18:
erase the board