tzapu/WiFiManager #1638: What is the maximum "found network list" on configuration page

Emcukcan, 2023-07-14:
I am using Wifi  Manager 2.0.4-beta, I have users which works in environment contains tens of routers and their problem, the router they want to use is not visible. It become visible only after they turn off some of other routers. 

Is there any maximum found network parameter or it is just not visible on the web page?
Thanks.

tablatronix, 2023-07-14:
hmm I used to test this with hundreds but then work turned off 2.5ghz.
There is probably a limit based on time scanning or memory. Are they all unique?

Can you get me a debuglog or sampling?

didier9, 2024-09-12:
This problem seems to still be present. The wm reports finding 6 networks but only 3 show up in the web page. Then the device does not actually connect to the AP:

*wm:Connecting to NEW AP: xxxxxxxxx
*wm:[ERROR] Connect to new AP Failed 
*wm:6 networks found

The device is an ESP8266 (HW-364B). I am not sure what additional information to provide.