tzapu/WiFiManager #1640: ESP32 - Autoconnect fails

lisander-lopez, 2023-08-03:
----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master (Added git url to platform.ini) (WiFiManager @ 2.0.16-rc.2+sha.3a303ee)

Esp8266/Esp32: ESP32

Hardware:	esp-wroom-32

Core Version:      	framework-arduinoespressif32 @ 3.20009.0 (2.0.9)

### Description

When connecting to my Router I get a "0x3007" error, according to the documentation it is `[ESP_ERR_WIFI_CONN](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#c.ESP_ERR_WIFI_CONN) (0x3007): WiFi internal control block of station or soft-AP error`

I've looked ALL OVER online (issues, ESP forum, et) I've tried stuff like: 

`WiFi.mode(WIFI_STA);` before `wm.autoConnect()`
`wm.setCleanConnect(true);` before autoConnect
 `setConnectTimeout`
`setTimeout`

My router is a Verizon CR1000B if it matters. It has an "iot" network feature that just locks an AP in 2.4ghz 

I would connect before, but I would have non blocking code that just tried to connect over and over and eventually it would work (sometimes in 5 mins sometimes in hours)

Any help would be appreciated thanks :) 
### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

```cpp
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

void setup() {
    Serial.begin(115200);
    WiFiManager wm;
    wm.setDebugOutput(true);
    wm.setConnectRetries(2);
    bool res;
    res = wm.autoConnect(); // auto generated AP name from chipid
    if(!res) {
        Serial.println("Failed to connect");
        ESP.restart();
    } 
    else {
        //if you get here you have connected to the WiFi    
        Serial.println("connected...yeey :)");
    }
}

void loop() {
    // put your main code here, to run repeatedly:   
}
```

### Debug Messages

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
*wm:AutoConnect 
*wm:Connect Wifi, ATTEMPT # 1 of 2
*wm:Connecting to SAVED AP: The Cloud
*wm:connectTimeout not set, ESP waitForConnectResult... 
*wm:Connect Wifi, ATTEMPT # 2 of 2
*wm:Connecting to SAVED AP: The Cloud
E (21294) wifi:sta is connecting, return error
[ 19488][E][WiFiSTA.cpp:317] begin(): connect failed! 0x3007
*wm:connectTimeout not set, ESP waitForConnectResult... 
*wm:AutoConnect: FAILED for  19446 ms
*wm:StartAP with SSID:  AutoConnectAP
*wm:AP IP address: 192.168.4.1
*wm:Starting Web Portal
```

tablatronix, 2023-08-03:
Try increasing setConnectionTimeout, i have never seen that error code before, maybe turn up ESP debug.
I bet there is some kind of error or bug in esp lib not being caught here, since there is not much info

lisander-lopez, 2023-08-04:
> Try increasing setConnectionTimeout, i have never seen that error code before, maybe turn up ESP debug. I bet there is some kind of error or bug in esp lib not being caught here, since there is not much info

Thank you for replying! I've been pulling my hair on this for a while.
I've changed the code to this (and turned on ESP debug)
```
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

void setup() {
    Serial.begin(115200);
    WiFiManager wm;
    WiFi.mode(WIFI_STA);
    wm.setDebugOutput(true);
   wm.setConnectTimeout(60 * 1); // 1 minute
    wm.setConnectRetries(2);
    bool res;
    res = wm.autoConnect(); // auto generated AP name from chipid
    if(!res) {
        Serial.println("Failed to connect");
        ESP.restart();
    } 
    else {
        //if you get here you have connected to the WiFi    
        Serial.println("connected...yeey :)");
    }
}

void loop() {
    // put your main code here, to run repeatedly:   
}
```

I put the output on a pastebin since it is too large/clunky to fit here https://pastebin.com/f0XtxChX
Im getting things like 
- `HANDSHAKE_TIMEOUT`
- `AUTH_EXPIRE`
- `ASSOC_EXPIRE`

you can also see that I have retries of 2 but it increases... super weird 😂

chriskinal, 2023-08-04:
Try taking the space out of the Wifi SSID. Maybe The-Cloud or The_Cloud versus "The Cloud". Try all lowercase and no spaces or other special characters. While spaces are technically allowed within the SSID, some implementations may not handle them well. Spaces are not permitted as the first or last character in an SSID.

Perhaps it is something with the Verizon router and/or the "IOT network" feature.  Have you tried a different 2.4GHz network or router? Can you add the MAC address of the ESP32 into the "allowed" list on the router? If using DHCP, can you set a reservation for the ESP32?

lisander-lopez, 2023-08-04:
> Try taking the space out of the Wifi SSID. Maybe The-Cloud or The_Cloud versus "The Cloud". Try all lowercase and no spaces or other special characters. While spaces are technically allowed within [...]

Thank you for contributing to this issue and trying to help.

I changed my SSID to `the-cloud`  and it worked the first time, but subsequent reboots the same error persisted.

I do have an option to create a "guest network" I tried that and no luck :( . I even temporarily shut off all bands expect for 2.4ghz on my main network and same issue. When the ESP32 first connected to my router it was "saved" so it knew its mac address etc. which I can then disallow if I wanted to (so by default its allow). I am using DHCP but I can give a static IP based on hostname and mac address, That didn't seem to fix my problem either.

chriskinal, 2023-08-04:
Try turning off encryption to see if it is a crypto negotiation problem. When you say "reboot", is that by a button on the dev board or disconnecting power? I have some boards that will connect once but when I push the button on the board it won't reconnect. When I disconnect and reconnect power it works. It behaves like this with my Linksys wifi. When I use the mobile hotspot feature on Windows 10 to create an AP it connects fine by pushing the button or power cycling.

tablatronix, 2023-08-05:
This is a huge issue with pmf and deauth protection I have an open issue with esp idf it has been a while since I tested this but not sure its fixed yet

lisander-lopez, 2023-08-05:
> Try turning off encryption to see if it is a crypto negotiation problem. When you say "reboot", is that by a button on the dev board or disconnecting power? I have some boards that will connect once [...]

Ok breaks to say it but it worked on an open network! I'm partially happy since I prefer to have a password on my AP.

lisander-lopez, 2023-08-05:
> This is a huge issue with pmf and deauth protection I have an open issue with esp idf it has been a while since I tested this but not sure its fixed yet

This one? https://github.com/espressif/esp-idf/issues/8192 Seems like they closed it, I am using the latest version too :/ 
```sh
> pio pkg update -g -p espressif32
Platform Manager: espressif32@6.3.2 is already up-to-date
Tool Manager: framework-arduinoespressif32@3.20009.0 is already up-to-date
Tool Manager: tool-esptoolpy@1.40501.0 is already up-to-date
Tool Manager: tool-mkfatfs@2.0.1 is already up-to-date
Tool Manager: tool-mklittlefs@1.203.210628 is already up-to-date
Tool Manager: tool-mkspiffs@2.230.0 is already up-to-date
Tool Manager: toolchain-xtensa-esp32@8.4.0+2021r2-patch5 is already up-to-date
```

tablatronix, 2023-08-05:
Can you confirm your IDF in code?

*DLOG: [1] ESP-IDF version:  4.4.5.230614

lisander-lopez, 2023-08-05:
> Can you confirm your IDF in code?
> 
> *DLOG: [1] ESP-IDF version: 4.4.5.230614

I used 
```
Serial.println("ESP-IDF version is: " + String(esp_get_idf_version()));
```

and got `ESP-IDF version is: v4.4.4`

So its not lastest, but it should include the "fix"

lisander-lopez, 2023-08-08:
So I can confirm that everything works smoothly when my WiFi is left insecure, but not working with WPA2