tzapu/WiFiManager #1645: autoConnect is blocking despite setConfigPortalBlocking(false)

entozoon, 2023-08-09:
Is it possible to run other things while autoConnect attempts to connect to wifi?
I don't want my device to just freeze up while it tries to connect.

### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32:  Esp32

### Description

Problem description
 
```cpp
#include <WiFiManager.h>
WiFiManager wm;
void setupWifi() {  
  wm.setConfigPortalBlocking(false);
  wm.setConnectTimeout(180);
  wm.setConfigPortalTimeout(180); 
  Serial.println("Wifi starting");
  while (!wm.autoConnect()) {
    Serial.println("?"); // No output
    delay(50);
  }
} 
void setup() {
  Serial.begin(115200);
  delay(3000);
  setupWifi();
} 
void loop() {
  wm.process();
  Serial.println("Loop");  
}
```

### Debug Messages

```
Wifi starting
*wm:AutoConnect 
*wm:Connecting to SAVED AP: My House
```

It hangs here while attempting to connect (which can take ages as I have a weak signal) but ideally I'd flash an LED or whatever while it tries.

tablatronix, 2023-08-10:
Yeah connection is blocking, there is another issue for this. I will have to add callbacks or something, you really cannot be doing anything in code when waiting for connection as it can affect it. I think I can add a return and a callback when connected or status change etc.

SiddiqueAhmad, 2023-09-16:
any chance ? We need to run load and battery lights, but it get stuck in connecting