tzapu/WiFiManager #1646: Autoreconnect doesn't work due to _arduino_event_cb(): STA Got Same IP

Dario-Ciceri, 2023-08-19:
[ 57622][V][WiFiGeneric.cpp:367] _arduino_event_cb(): STA Got Same IP:192.168.173.158
11:35:48.219 -> Got time adjustment from NTP!
11:35:49.962 -> *wm:AutoConnect: FAILED 

Hi, I noticed that even if ESP32 connects to WiFi, switching from ethernet module, wifi manager will say *wm:AutoConnect: FAILED 
but ESP32 is successfully connected...

tablatronix, 2023-08-19:
Interesting

Dario-Ciceri, 2023-08-19:
I don't know if there's a way to "reset" WiFi and avoid STA Got Same IP by erasing the previous one... I tried many esp API calls but nothing works, even deinit and init again doesn't work...
If I reset esp32 it connects without problem and I think it's because it doesn't have "a previous ip"

tablatronix, 2023-08-19:
Sounds like it might be an ESP lib issue, either way I will see if we can workaround it

Dario-Ciceri, 2023-08-19:
> Sounds like it might be an ESP lib issue, either way I will see if we can workaround it
> 
> is [ 57622][V][WiFiGeneric.cpp:367] _arduino_event_cb(): STA Got Same IP:192.168.173.158 the only event you see?
> 
> I wonder if switching ap to ap would also do this?

Yes and routine stuff like STA disconnected, start etc due to wifimanager behaviour

Dario-Ciceri, 2023-08-19:
> > Sounds like it might be an ESP lib issue, either way I will see if we can workaround it
> > is [ 57622][V][WiFiGeneric.cpp:367] _arduino_event_cb(): STA Got Same IP:192.168.173.158 the only event you see?
> > I wonder if switching ap to ap would also do this?
> 
> Yes and routine stuff like STA disconnected, start etc due to wifimanager behaviour
> 
> Switching from ap to ap? Idk, honestly it would be weird if different AP same IP no?

oh and also, sometimes a weird error which seems related to `  // if retry without delay (via begin()), the IDF is still busy even after returning status
  // E (5130) wifi:sta is connecting, return error
  // [E][WiFiSTA.cpp:221] begin(): connect failed!
` but even with `if (_aggresiveReconn) delay(1000);` sometimes I get `// E (5130) wifi:sta is connecting, return error
  // [E][WiFiSTA.cpp:221] begin(): connect failed!`

Dario-Ciceri, 2023-08-19:
could you also tell me what's causing hostByName(): DNS Failed for www.google.com ?

tablatronix, 2023-08-20:
What esp version are you using ?

Dario-Ciceri, 2023-08-30:
ehy @tablatronix I think I found the problem, look here please: https://github.com/espressif/arduino-esp32/issues/653#issuecomment-755851929