tzapu/WiFiManager #1649: WiFi Scan ASYNC not always works

Dario-Ciceri, 2023-08-28:
On esp32 WiFi Scan ASYNC started and then nothing... sometimes it works after 6 seconds or 5...

Dario-Ciceri, 2023-08-28:
I searched for `WiFi.scanNetworks` and found this one: `res = WiFi.scanNetworks(true);` which is weird ... (?)

Why using `true`? 
These are the possible overrides: 

1.     bool          WiFi_scanNetworks(); 
2.     bool          WiFi_scanNetworks(bool force, bool async);
3.     bool          WiFi_scanNetworks(unsigned int cachetime, bool async);
4.     bool          WiFi_scanNetworks(unsigned int cachetime);

Is it intentional to use `WiFi_scanNetworks(unsigned int cachetime);` with `cachetime = true = 1` ????

georgevbsantiago, 2023-08-29:
I also identified that the ESP32 (devkit with Platformio) is taking time to map the Routers signal. And it doesn't always work. Sometimes I have to click the Refresh button for it to work.

tablatronix, 2023-08-31:
ESP WiFi.scanNetworks(true);
WM WiFi_scanNetworks(true);

are 2 different things

Dario-Ciceri, 2023-08-31:
> ESP WiFi.scanNetworks(true); WM WiFi_scanNetworks(true);
> 
> are 2 different things, not overrides

uhm ok, I checked these days and tried to understand, I changed to match async call and it works

tablatronix, 2023-09-16:
I am not sure this is the async scanning that is the issue, I am seeing 0 scans also sometimes.

you can go into the .h and change this to false to disable it

    boolean       _asyncScan              = true; // perform wifi network scan async

closed 2023-11-04 by tablatronix