tzapu/WiFiManager #1650: How to disable update and info pages?

georgevbsantiago, 2023-08-29:
Hello,

I limited the display of the menus with the method:
```
const char* menu[] = {"wifi","param","sep","exit"};
wifiManager.setMenu(menu, 4);
```

However, when we typed "192.168.4.1/update" and "192.168.4.1/info", we were able to access the HTML pages.

Is there a way to disable the "update" and "info" pages so that the user cannot access them through the browser?

georgevbsantiago, 2023-08-29:
In 'wm_consts_en.h', I changed the values to "". Apparently it worked. But I don't know if this is correct or can cause bugs.

```
const char R_info[]               PROGMEM = "";
const char R_update[]             PROGMEM = "";
```

tablatronix, 2023-09-08:
That works, you can also add overrides to those routes in user code

tablatronix, 2023-09-08:
```
}

void bindServerCallback(){
  wm.server->on("/custom",handleRoute);

  // you can override wm route endpoints
  wm.server->on("/info",handleNotFound);
  wm.server->on("/update",handleNotFound);
  wm.server->on("/erase",handleNotFound);
}
```

I am not sure there is a way to remove webserver handlers, maybe an empty lambda, but I had crashes with my preliminary tests.

I hope to add these features as build options but for now ust use your own handler for these, I will probably also make wm.handleNotFound public

tablatronix, 2023-09-08:
eb27c525e33e6ce558bd27496685ec657f2cb23e

closed 2023-09-08 by tablatronix

tablatronix, 2023-09-08:
#982