tzapu/WiFiManager #1659: Custom Parameter dont show

woodenplastic, 2023-09-26:
Hi,

When i use more than 20 custom Parameter the captive Portal doesnt work anymore and i have to refresh the Seite endlessly until the parameters randomly appear.

Is there any limits for parameters i have to keep in mind?

tablatronix, 2023-09-29:
you are running out of memory, either in param storage or in the html string flush, There are some things we can do to reduce that but will require code changes

woodenplastic, 2023-09-30:
i searched some other issues and i saw other people having this problem if i use a lot of custom html parameters. i saw branches with chunked respones but im not able to get it done because i lost oversight over all needed changes.  is there a way you could show me how to do it?

tablatronix, 2023-09-30:
Yeah no-one ever tested those, I will try to  update them, I intended to add a dynamic param test to add a bunch to test memory, and these fixes and never got around to it. There is also the option to multipart flush the webserver output, not sure how though

woodenplastic, 2023-09-30:
That would be lovely! I will do testing if you got something. 

I am working on a example to show checkboxes and sliders with easy styling to contribute something. It seems to be a problem for many people. 
With the css updated this library is a capable solution for a lot of usecases.

tablatronix, 2023-09-30:
Looks like there is a tooltip class already

<span id="sliderLabel0" class="tooltiptext">Effect speed</span>

woodenplastic, 2023-10-03:
void WiFiManager::HTTPSend(const String &content) {
  server->sendHeader("Transfer-Encoding", "chunked");
  server->send(200, "text/html", "");  // Send the initial HTTP headers

  size_t contentLength = content.length();
  size_t chunkSize = 256;  // Adjust this value based on your requirements

  for (size_t i = 0; i < contentLength; i += chunkSize) {
    size_t chunkEnd = min(i + chunkSize, contentLength);
    size_t chunkLength = chunkEnd - i;

    // Send the chunk size in hexadecimal followed by CRLF
    server->sendContent(String(chunkLength, HEX));
    server->sendContent("\r\n");

    // Send the actual chunk data
    server->sendContent(content.substring(i, chunkEnd));
    server->sendContent("\r\n");
  }

  // Send a zero-length chunk to indicate the end of the response
  server->sendContent("0\r\n\r\n");
}

that helped a lot for me, parameters show now consistenly up in wifi (interestingly not on the first load of the page in wifinoscan)

tablatronix, 2025-06-29:
I have a test branch for this and didnt get very far with testing it, Ill take a look at what you have created