tzapu/WiFiManager #1663: wm.autoConnect(ssid, password); consumes cumulative memory

RulioHT2021, 2023-10-09:
## PLEASE TRY Latest Master BRANCH before submitting bugs, in case they were already fixed. ##

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32: 

Hardware:			ESP-12e, esp01, esp25

Core Version:      	2.4.0, staging

### Description

Problem description

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch
void task_reconnect(void * parameter){
  task_rec = true;
  wm.autoConnect(ssid, password);
  if(WiFi.status() == WL_CONNECTED){
    Serial.println("WIFI OK");
    timeClient.update();
    ping_test();
    Serial.println(resp);
    if(client.connected()){
      Serial.println("BROKER OK.");
    }else if(client.connect("A")){
      Serial.println("BROKER OK.");
    }else{
      Serial.println("ERRO BROKER não conectado.");
    }
  }else{
    Serial.println("ERRO WIFI não conectado");
  }
  Serial.print("ESP.getFreeHeap(): ");
  Serial.println(ESP.getFreeHeap());
  vTaskDelete(NULL);    // Delete task
}
#END
```

### Debug Messages

```
messages here
```

RulioHT2021, 2023-10-09:
Sorry, I'm new here, I'm having a memory problem, as each reconnection test consumes more memory.

tablatronix, 2023-10-10:
Will test thanks

tablatronix, 2023-10-10:
I am assuming you have removed your local callouts to timeClient, ping, client checks and you are not leaking memory in them ?

tablatronix, 2023-10-10:
I am calling autoconnect in a loop and not leaking any memory.. Are you using blocking or non blocking mode?

RulioHT2021, 2023-10-10:
Sorry I wasn't so clear, after connecting the network I turn off the router and with each reconnection attempt with the router turned off it consumes memory.
I'm using it without blocking.

tablatronix, 2023-10-11:
Ok so the configportal is starting each time? How do you get out of it, do you have a timeout?

tablatronix, 2023-10-11:
Not seeing it, what exact hardware you using ?

RulioHT2021, 2023-10-11:
I make a routine to check if 60 seconds have passed or if the timestamp is correct when it is 50 seconds, and I leave it inside the main loop.

tablatronix, 2023-10-11:
Are you sure your tasks are cleaned up after each ?

RulioHT2021, 2023-10-11:
I'll need to test other routines, I'll get back to you as soon as possible.

RulioHT2021, 2023-10-11:
Tanks.

RulioHT2021, 2023-10-19:
16:38:35.036 -> ESP.getFreeHeap(): 207204
16:38:35.036 -> ERRO WIFI não conectado
16:38:37.090 -> Chamado envio MQTT.
16:38:37.090 -> ESP.getFreeHeap(): 218284
16:38:37.090 -> Timestamp incorreto.

`void task_reconnect(void * parameter){
  Serial.print("ESP.getFreeHeap(): ");
  Serial.println(ESP.getFreeHeap());
  wm.autoConnect(ssid, password);
  Serial.print("ESP.getFreeHeap(): ");
  Serial.println(ESP.getFreeHeap());
  if(WiFi.status() == WL_CONNECTED){
    Serial.println("WIFI OK");
    timeClient.update();
    ping_test();
    Serial.println(resp);
    if(client.connected()){
      Serial.println("BROKER OK.");
    }else if(client.connect((char*) s_id.c_str(), mqttUser, mqttPassword)){
      Serial.println("BROKER OK.");
    }else{
      Serial.println("ERRO BROKER não conectado.");
    }
  }else{
    Serial.println("ERRO WIFI não conectado");
  }
  vTaskDelete(NULL);    // Delete task
}`

RulioHT2021, 2023-10-19:
I created a task to reconnect and as you can see, each call consumes a little memory after wm.autoConnect(ssid, password);
The scenario I'm testing is:
in case there is a power outage and the router does not come back on.
Sooner or later it will restart due to lack of memory.
The problem is that I'm running a routine that cannot be restarted.

tablatronix, 2023-10-19:
Just set WM debug level to WM_DEBUG_MAX it will dump heap stats

tablatronix, 2023-10-20:
Looks like leaking 1440 bytes, hmm

tablatronix, 2023-10-20:
I cant find it yet, but my memory jumps up and down but stays <100% so something is cleaning up, but maybe not under certain conditions. I will have to run memory deugging and find what on the heap. I thought It was the known DNSD issue but that might only be a tiny bit.

Do you have many AP found during scans?
Are you using hostname or mdns?
If you change the configportaltimeout to like 20 seconds does anything change memory wise?

RulioHT2021, 2023-10-20:
partitions: LITTLEFS

If you change the configportaltimeout to like 20 seconds does anything change memory wise?
yes, Do not change

Are you using hostname or mdns?
no

RulioHT2021, 2023-10-20:
`
void task_reconnect(void * parameter){
  Serial.print("ESP.getFreeHeap(): ");
  Serial.println(ESP.getFreeHeap());
  WiFi.begin();
  timeClient.begin();
  //wm.autoConnect(ssid, password);
  Serial.print("ESP.getFreeHeap(): ");
  Serial.println(ESP.getFreeHeap());
  if(WiFi.status() == WL_CONNECTED){
    Serial.println("WIFI OK");
    timeClient.update();
    ping_test();
    Serial.println(resp);
    if(client.connected()){
      Serial.println("BROKER OK.");
      timeClient.begin();
    }if(client.connect((char*) s_id.c_str(), mqttUser, mqttPassword)){
      Serial.println("BROKER RECONECTADO.");
    }else{
      Serial.println("ERRO BROKER não RECONECTADO.");
    }
  }else{
    Serial.println("ERRO WIFI não conectado");
  }
  vTaskDelete(NULL);    // Delete task
}
`

tablatronix, 2023-10-20:
Yeah I am not really sure what the issue is.. I will keep testing

My memory does this if i autoconnect loop
[image]