tzapu/WiFiManager #1664: reset when updating the website and possibility of removing adding options

serlancelot, 2023-10-10:
Hello! good morning!
I discovered this library that solves a problem for me, I created a graphical configuration environment with the ESPUI library that already had a WiFi configuration section in the example, it turns out that it fell short (since I couldn't do things that with HTML, CSS and JS allow) so the wifi part didn't work for me until I found this library.

After this mini introduction... I'll tell you the problem I have, well at the moment there are two.

The first is that: once the WiFi is configured, I connect to the ESP's IP, it loads the website and I ask it to refresh the website, it disconnects and the ESP restarts, so it puts it on the serial monitor (previously I had it configured with the wifi and key in the code and it did not pass). Why does this happen? This only happens if I update with the mobile with the computer, nothing happens

And the second thing: it is possible to eliminate some option from the wifimanager menu (that only scan wifi appears or manually configure wifi) as well as if it is possible to add an option to be able to add and remove the AP (in case you are in a place without wifi to create the AP point and then load the website I have)

Thank you very much in advance!! all the best!

Arduino IDE 2.1.0
ESP32 dev module
ESP32 de Espressif 2.0.14
WiFiManager 2.0.16
LittleFS_esp32 1.0.5 If I update it to 1.0.6 (the website does not work)
ESPAsyncWebServer 1.2.6

```ruby
#include <WiFiManager.h>
#include <strings_en.h>
#include <wm_consts_en.h>
#include <wm_strings_en.h>
#include <wm_strings_es.h>
#include "WiFi.h"
#include "ESPAsyncWebServer.h"
#include <LITTLEFS.h>

byte selectorwifi = true;

// Create AsyncWebServer object on port 80
AsyncWebServer server(80);

void setup(){
  
  Serial.begin(115200);
  // Initialize LITTLEFS
  if(!LITTLEFS.begin(true)){
    Serial.println("An Error has occurred while mounting LITTLEFS");
    return;
  }
if(selectorwifi==false){ 
      WiFi.mode(WIFI_AP);
  while(!WiFi.softAP("Test", NULL))
     {
       Serial.println(".");
       delay(100);
      }
   }
  else{
    WiFiManager wm;
     //wm.resetSettings();
    bool res;
    res = wm.autoConnect("AutoConnectAP",NULL);
  }
server.serveStatic("/", LITTLEFS, "/");
server.serveStatic("/", LITTLEFS, "/").setDefaultFile("/index.html");

  // Route to load style.css file
server.on("/normalize.min.css", HTTP_GET, [](AsyncWebServerRequest *request){
   request->send(LITTLEFS, "/normalize.min.css", "text/css");
  });

  // Start server
  server.begin();
  LITTLEFS.begin();
}
 
void loop(){
 
}
```

```ruby
12:42:58.351 -> E (2043310) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
12:42:58.351 -> E (2043310) task_wdt:  - async_tcp (CPU 0/1)
12:42:58.351 -> E (2043310) task_wdt: Tasks currently running:
12:42:58.396 -> E (2043310) task_wdt: CPU 0: IDLE
12:42:58.396 -> E (2043310) task_wdt: CPU 1: loopTask
12:42:58.396 -> E (2043310) task_wdt: Aborting.
12:42:58.396 -> 
12:42:58.396 -> abort() was called at PC 0x400f9e94 on core 0
12:42:58.396 -> 
12:42:58.396 -> 
12:42:58.396 -> Backtrace: 0x40083691:0x3ffbec8c |<-CORRUPTED
12:42:58.396 -> 
12:42:58.396 -> 
12:42:58.396 -> 
12:42:58.396 -> 
12:42:58.396 -> ELF file SHA256: f5053ececed9f6d2
12:42:58.396 -> 
12:42:58.708 -> Rebooting...
12:42:58.708 -> ets Jul 29 2019 12:21:46
12:42:58.708 -> 
12:42:58.708 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
12:42:58.708 -> configsip: 0, SPIWP:0xee
12:42:58.708 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
12:42:58.708 -> mode:DIO, clock div:1
12:42:58.708 -> load:0x3fff0030,len:1344
12:42:58.708 -> load:0x40078000,len:13964
12:42:58.708 -> load:0x40080400,len:3600
12:42:58.708 -> entry 0x400805f0
12:42:59.108 -> *wm:AutoConnect 
12:42:59.201 -> *wm:Connecting to SAVED AP: vodafoneCA8B
12:42:59.712 -> *wm:connectTimeout not set, ESP waitForConnectResult... 
12:43:02.314 -> *wm:AutoConnect: SUCCESS 
12:43:02.314 -> *wm:STA IP Address: 192.168.0.25

////////////////////////////////////////////////////////////////////////////////
I update the website from my mobile
///////////////////////////////////////////////////////////////////////////////
This only happens if I update with the mobile with the computer, nothing happens
////////////////////////////////////////////////////////////////////////////////////////////

12:45:09.792 -> E (136472) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
12:45:09.792 -> E (136472) task_wdt:  - async_tcp (CPU 0/1)
12:45:09.792 -> E (136472) task_wdt: Tasks currently running:
12:45:09.837 -> E (136472) task_wdt: CPU 0: IDLE
12:45:09.837 -> E (136472) task_wdt: CPU 1: loopTask
12:45:09.837 -> E (136472) task_wdt: Aborting.
12:45:09.837 -> 
12:45:09.837 -> abort() was called at PC 0x400f9e94 on core 0
12:45:09.837 -> 
12:45:09.837 -> 
12:45:09.837 -> Backtrace: 0x40083691:0x3ffbec8c |<-CORRUPTED
12:45:09.837 -> 
12:45:09.837 -> 
12:45:09.837 -> 
12:45:09.837 -> 
12:45:09.837 -> ELF file SHA256: f5053ececed9f6d2
12:45:09.837 -> 
12:45:10.006 -> Rebooting...
12:45:10.006 -> ets Jul 29 2019 12:21:46
12:45:10.006 -> 
12:45:10.006 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
12:45:10.006 -> configsip: 0, SPIWP:0xee
12:45:10.006 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
12:45:10.006 -> mode:DIO, clock div:1
12:45:10.052 -> load:0x3fff0030,len:1344
12:45:10.052 -> load:0x40078000,len:13964
12:45:10.052 -> load:0x40080400,len:3600
12:45:10.052 -> entry 0x400805f0
12:45:10.422 -> *wm:AutoConnect 
12:45:10.515 -> *wm:Connecting to SAVED AP: vodafoneCA8B
12:45:11.012 -> *wm:connectTimeout not set, ESP waitForConnectResult... 
12:45:13.608 -> *wm:AutoConnect: SUCCESS 
12:45:13.608 -> *wm:STA IP Address: 192.168.0.25
```

tablatronix, 2023-10-10:
Can you try the git version?

There have been issues running async and webserver libraries together, I am suprised it works at all, also there is clearly a watchdog problem somewhere..

tablatronix, 2023-10-10:
For menu config see `setMenu`

serlancelot, 2023-10-10:
Hello tablatronix! 
Forgive my ignorance, what do you mean by the git version?
thx

tablatronix, 2023-10-10:
Development version from master

serlancelot, 2023-10-10:
Hello again. I already have the menu, the git... I have version v2.0.16-rc.2 which is the last one that appears here and in the Arduino library manager.
I don't know where to get the "master" from. Again, sorry for the ignorance.

serlancelot, 2024-02-23:
Hello, for various reasons I left the project "paused" and now I have started with it again. I can't get the connection to remain stable when I use the ESP as an independent point, I have tried it with Android and a cell phone with iOS but it does the same thing... Whether connected to Wi-Fi or independent, as soon as you connect with a cell phone it disconnects and reconnect.... How could I solve it?

serlancelot, 2024-02-23:
It seems that the problem is with the images on the website... I have removed them all and there are no problems or so it seems... I'm going to keep looking.

tablatronix, 2024-02-24:
Could be maxing out heap when sending strings. Would need some memory buffer to send smaller content but without chunked responses its a pain

serlancelot, 2024-02-24:
But what I don't understand is that with the computer there are no problems and with mobile devices there are problems... The truth is that the gifs were compressed through a website so that they would take up less memory in the ESP and with the belief that the same as the compressing takes up less space and would load faster... it may be doing the opposite of what I wanted... 
at least the problem is apparently located...