tzapu/WiFiManager #1666: Wi-Fi connection status via LED

serlancelot, 2023-10-11:
Hello, I have tried to put an LED to indicate if it is connected to the Wi-Fi and it works halfway.
I tell you:
I have set two "modes", one would be an AP with which the LEDs already work (in a somewhat sloppy way) and the other would be through Wifimanager (which creates the "temporary" AP to connect to the router's Wifi ).
This is the one where I have problems getting one LED to light up when it is not connected and another when it is.
The problem is the following, when you connect to the router the status LED changes fine but when you turn off the router it still has the active connection LED and when the temporary AP is created it does nothing (I don't know how to do it).
Thanks greetings!

Arduino IDE 2.1.0
ESP32 dev module
ESP32 de Espressif 2.0.14
WiFiManager 2.0.16
LittleFS_esp32 1.0.5 If I update it to 1.0.6 (the website does not work)
ESPAsyncWebServer 1.2.6

```ruby
#include <WiFiManager.h>
#include <strings_en.h>
#include <wm_consts_en.h>
#include <wm_strings_en.h>
#include <wm_strings_es.h>
#include "WiFi.h"
#include "ESPAsyncWebServer.h"
#include <LITTLEFS.h>

int wifi_led_on = 23;
int wifi_led_off = 22;
int connections;

byte selectorwifi = true;

// Create AsyncWebServer object on port 80
AsyncWebServer server(80);

void setup() {
  pinMode(wifi_led_on, OUTPUT);
  pinMode(wifi_led_off, OUTPUT);
  Serial.begin(115200);

  // Initialize LITTLEFS
  if (!LITTLEFS.begin(true)) {
    Serial.println("An Error has occurred while mounting LITTLEFS");
    return;
  } 
      if (WiFi.status() !=WL_CONNECTED) {
      digitalWrite(wifi_led_on, LOW);
      digitalWrite(wifi_led_off, HIGH);   
  }
  if (selectorwifi == false) {
    WiFi.mode(WIFI_AP);
    while (!WiFi.softAP("Test", NULL)) {

      Serial.println(".");
      delay(100);
    }
  }
  if (selectorwifi == true) {

    WiFiManager wm;
    wm.setDarkMode(true);
    std::vector<const char *> menu = { "wifi", "wifinoscan", "sep", "erase" };
    wm.setMenu(menu);  // custom menu, pass vector
    //wm.resetSettings();
    bool res;
    res = wm.autoConnect("Test", NULL);
  }

  server.serveStatic("/", LITTLEFS, "/");
  server.serveStatic("/", LITTLEFS, "/").setDefaultFile("/index.html");

// Route to load style.css file
  server.on("/normalize.min.css", HTTP_GET, [](AsyncWebServerRequest *request) {
    request->send(LITTLEFS, "/normalize.min.css", "text/css");
  });

  // Start server
  server.begin();
  
  if (WiFi.status() == WL_CONNECTED) {
      digitalWrite(wifi_led_off, LOW);
      digitalWrite(wifi_led_on, HIGH);
  }
  
  LITTLEFS.begin();
}

void loop() { 
connections=WiFi.softAPgetStationNum();
if(selectorwifi==false){
  if(connections>0){
      digitalWrite(wifi_led_off, LOW);
      digitalWrite(wifi_led_on, HIGH);
  }

  if(connections==0){
      digitalWrite(wifi_led_off, HIGH);
      digitalWrite(wifi_led_on, LOW);
  }
}
}
```

tablatronix, 2023-10-11:
Use non blocking mode

closed 2023-10-11 by tablatronix

serlancelot, 2023-10-11:
Hello, thanks for the response, I have managed to solve almost everything... I still have the problem with the AP that Wifimanager generates, how is there a WiFi.softAPgetStationNum() style command? so that when someone connects I can turn on the LED... (like I have with WiFi.mode(WIFI_AP);)
Thanks greetings!