tzapu/WiFiManager #1667: Feature request : connect through ip

kamikazem, 2023-10-12:
In the projects that connect to the internet but at the same time offer some kind of web page for control, Retrieving ip and connecting to the device control panel is cumbersome for non technical users mostly because they cannot find the ip address easily.

So When a device is connected to a wifi with sta , it might be useful to keep the ap mode and show list of wifis.
But also show info of the connected wifi and a link to the user.

 when this link , for example http://192.168.200.12@theSsid gets clicked the following set of commands get executed:
1.add this address to favorites with a custom name
2.execute the wifi credentials links
3.deauth the current user that is connected through ap
4.lower ap transmission power lower than that wifi with TheSsid
5.redirect to that link (without the @TheSsid) with a 10 second delay

I think In this manner The ip address finding issue would be easily solved for non technical users
I don't know if you can set the transmission power of the device separately for ap and sta mode, if you could the hotspot could be always run with a very low transmission power for next possible interactions.

tablatronix, 2023-10-12:
You could always use mdns

kamikazem, 2023-10-13:
According to my experience, mdns does not always work, for example when the client is on a vpn.

The user doesn't know the ip address , because he is not tech savvy and the ip is assigned by dhcp of the router.

But esp32 knows it because it's connected to the network in station mode through that ip address.

tablatronix, 2023-10-13:
hmm, I think this can all be done already in code.
Let me test and make an example

tablatronix, 2023-11-18:
This already exists
[image]

Miceuz, 2023-11-29:
Hi @tablatronix ,
Can you provide short guidelines on how did you achieve that? I have this same task now - provide IP address to the user after connecting to WiFi network. As I understand, after connecting to WiFi network, AP of WiFiManager is no longer running, or am I wrong?

kamikazem, 2023-12-12:
Hi @tablatronix 
I guess steps that i mentioned are doable in a captive portal, however since English is my second language i might not be able to communicate it well , but
Step 1,2 and 5 are doable with JavaScript at the client side and 3 and 4 should be doable on the arduino side.

tablatronix, 2023-12-12:
You can start the configportal and stay connected to wifi at the same time. It sometimes is unstable. But if you start configportal ondemand when connected to wifi it will still be online