tzapu/WiFiManager #1672: Feature request: Make _preloadwifiscan publicly settable

realA10001986, 2023-11-02:
In my projects, the WiFi scan triggered when starting the CP disturbs required network traffic between devices, especially during boot. Setting _preloadwifiscan to false does avoid this, but I would like avoid my users having to patch the WiFiManager library source code when compiling my projects. 

Adding a simple, public setPreLoadWiFi(bool) would really help.

tablatronix, 2023-11-03:
Yeah I am about to turn that off by default it was experimental and does make the page come up faster, but can also be problematic when joining or using ondemand and trying to join immediatly.

realA10001986, 2023-11-03:
OK, plain off is fine, too. Thanks!

tablatronix, 2023-11-04:
94529cd08b5d956f3ca015bb4b6fd0226ee43143

closed 2023-11-04 by tablatronix