tzapu/WiFiManager #1674: Issue with ESP32-S3 Using Wi-Fi on Demand

Fishbone69, 2023-11-13:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:   2.0.16-rc.2

Esp8266/Esp32: 

Hardware:			esp32-S3
Core Version:      	2.4.0, staging

### Description

I am trying to use WiFiManager On Demand to go out and grab the time from ntp server.  The following code worked great on Wemos esp32-C3 Pico but when I ported to Wemos esp32-S3 mini, it resets the chip at various places in the subroutine below.  It also resets the chip on a second, custom esp32-S3 dev board I used - so I think it is an issue with esp32-S3.  I am  fairly new to programming microcontrollers so I'll do my best to comply with any requests for additional data on this issue.   Also, I am using the PlatformIO IDE.  Any help is appreciated!

### Settings in IDE

Module:  			Wemos esp32-S3 Mini

Additional libraries: 

### Sketch
When a certain button is pressed the code executes the following subroutine.  Including the full sketch is impossible 

#BEGIN
#include <Arduino.h>
#include <WiFiManager.h>
//This is the subroutine
void startWiFi(){
  int wifiTimeout = 180;
  const char* ntpServer = "pool.ntp.org";
  const long utcOffsetSeconds = 3600*utcOffset;
  sntp_sync_status_t syncStatus;
  u8g2.clearBuffer();
  u8g2.setFontMode(1);
  u8g2.setDrawColor(2);
  WiFi.mode(WIFI_STA); //AFTER THIS LINE IS EXECUTED, THE CHIP WILL RESET BUT SOMETIMES GETS FURTHER AND MAY PRINT SOME OF THE MESSAGES BELOW.  IT WILL NEVER CONNECT TO WI-FI
  WiFiManager wm;
  wm.resetSettings();
  u8g2.setFont(u8g2_font_freedomSuperTiny_mf);
  u8g2.setCursor(0, 15);
  u8g2.print("STARTING WIFI...");
  u8g2.sendBuffer();
  wm.setConfigPortalTimeout(wifiTimeout); 
  u8g2.setCursor(0, 30);
  u8g2.print("SELECT \"ENCOM DW 1.0\"...");
  u8g2.sendBuffer();
  if (!wm.startConfigPortal("ENCOM DW 1.0")) {
    u8g2.setCursor(0, 45);
    u8g2.print("FAILED. WIFI TIMEOUT");
    u8g2.sendBuffer();
    delay(3000);
  }
  else {
    u8g2.setCursor(0,45);
    u8g2.print("CONNECTED.");
    u8g2.setCursor(0,60);
    u8g2.print("IP "+ WiFi.localIP().toString());
    u8g2.sendBuffer();
    configTime(utcOffsetSeconds, 3600*dstFlag, ntpServer);
    syncStatus = sntp_get_sync_status();
    while (syncStatus != SNTP_SYNC_STATUS_COMPLETED) {
        syncStatus = sntp_get_sync_status();
        delay(100); // Adjust the delay time as per your requirements
    }
    rtcDS3231.adjust(DateTime(static_cast<uint16_t>(rtcESP32.getYear()), static_cast<uint8_t>(rtcESP32.getMonth()+1), static_cast<uint8_t>(rtcESP32.getDay()), static_cast<uint8_t>(rtcESP32.getHour(true)), static_cast<uint8_t>(rtcESP32.getMinute()),static_cast<uint8_t>(rtcESP32.getSecond())));
    sntp_stop();  //resets syncStatus
    u8g2.clearBuffer();
    u8g2.setFont(u8g2_font_freedomsmall_mf);  
    u8g2.setCursor(64-11*6/2,30);
    u8g2.print("TIME  SET!");
    u8g2.drawFrame(15,15,98,20); 
    u8g2.drawFrame(17,17,94,16); 
    u8g2.sendBuffer();
    messageTone();
    delay (4000);
    lastButtonPressedTime = millis();
  }
  WiFi.disconnect(true);
  WiFi.mode(WIFI_OFF);
}  
#END
```

### Debug Messages
No messages, the chip just resets.  If there is a way to  
```
messages here
```

tablatronix, 2023-11-13:
There has to be serial exception no esp just resets itself. Try to get logs and maybe turn up esp debugging. I use the S3 all the time

tablatronix, 2023-11-13:
You would have to reduce this sketch alot, how are you calling this from button press?

Fishbone69, 2023-11-13:
@tablatronix It was surprising to me as well - especially since it worked perfectly on the esp32-C3.   I'll see if I can get logs - like I said, I new to programming microcontrollers but I'll try to figure it out.  As far as calling the code from the button press, I am using button2 library which uses callback functions.  So I use the following code:
```
buttonEnter.setPressedHandler([](Button2 &btn){
            pressed(dummyButton);
            if (currentSetSave == "YES"){
            dstFlag=setDST;
            utcOffset = setUtcOffset;
            startWiFi();
            mode = 1;
            settingsMode = 0;
            setTimeMode=0;
            selectSettingField = 0;
            }
            else{
            selectSettingField = 0;  
            }
           });
```
However, I keep coming back to the fact that this code works on my esp32-C3 board.

I'll get you logs if I can - Thanks!

Fishbone69, 2023-11-13:
I was trying to see if I could get info on the serial monitor and was actually able to run through the entire code in one instance (yes, it set the correct time from the server) although I made no significant changes to the code - just some Serial.println statements.   I was not able to get it to repeat that again, but if I remove power and repeat the button press, it seems to make it to different points in the startWiFi() subroutine before the esp32 restarts. 

At first I was wondering if the devboard could have an emi issue with the WiFi causing a reset.  But I don't think I would see that in 2 different esp32-S3 designs.

Fishbone69, 2023-11-13:
If I put the esp32-S3 in deep sleep then awaken it between attempts, I can get to the point where I can see the esp32-S3 on my android device - but upon trying to connect, it resets.  Subsequent attempts cause it to reset immediately.  Almost like an initialization issue.  I dunno.

tablatronix, 2023-11-16:
Is this `buttonEnter.setPressedHandler([]`  firing inside an interrupt?

Fishbone69, 2023-11-16:
No, that is not 

> Is this `buttonEnter.setPressedHandler([]` firing inside an interrupt?

No.  That is not inside an interrupt.  I am still looking into this but got sidetracked a bit.

Fishbone69, 2023-11-18:
I apologize to this community for bothering you.  It is a known problem with this series of boards:
[https://esp32.com/viewtopic.php?t=28506](url)
I will try to do more homework next time before starting an issue.  

Again, apologies.

closed 2023-11-18 by Fishbone69

tablatronix, 2023-11-18:
#1478