tzapu/WiFiManager #1678: Feature request: connect with specific BSSID obtaining from of a networkscann

lastphoenx, 2023-11-18:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master 2.016-rc.2

Esp8266/Esp32: 

Hardware:			ESP32CAM AI Thinker ESP32-Cam

### Description

Problem description

it would be very useful  to be able to do soemthing like this, after a networkscan:

WiFi.begin(ssid, password, Kanal, WiFi.BSSID(i))

Message:
As I found, that it often reconnects to first BSSID it has ever connectet, it tends to only reconnect to this BSSID.
with WiFi.begin(ssid, password, Kanal, WiFi.BSSID(i)) i can change that behavior easily, but I cant do that with Wifi.Manger as I dont have access to the password for example and as i dont wanna store it in my coding I use Wifi-Manger.

tablatronix, 2023-11-18:
This is based on the scan options esp does this because its faster but sucks for mesh. You can adjust this and you can also save bssid automatically. 
I will find examples

lastphoenx, 2023-11-18:
@tablatronix 
okay, I'd be really grateful for examples because I couldn't figure it out with custom parameters/add parameters.

tablatronix, 2023-11-18:
// Set ESP WIFI specifics 
  // WiFi.setScanMethod(WIFI_ALL_CHANNEL_SCAN); // wifi_scan_method_t scanMethod [WIFI_FAST_SCAN],WIFI_ALL_CHANNEL_SCAN
  // WiFi.setSortMethod(WIFI_CONNECT_AP_BY_SIGNAL); // wifi_sort_method_t sortMethod - [WIFI_CONNECT_AP_BY_SIGNAL],WIFI_CONNECT_AP_BY_SECURITY
  // WiFi.setMinSecurity(WIFI_AUTH_WPA2_PSK); [WIFI_AUTH_WPA2_PSK]

check the super example

lastphoenx, 2023-11-18:
wow, looks brilliant. thx so much.
meanwhile I tried this:

`
        #include <EEPROM.h>
        #include <WiFi.h>
        
        // Definiere die Adresse im EEPROM, an der die BSSID gespeichert wird
        #define BSSID_EEPROM_ADDRESS 0
        #define BSSID_LENGTH 6
        
        void saveBSSID(uint8_t *bssid) {
          // Speichere die BSSID in EEPROM
          for (int i = 0; i < BSSID_LENGTH; ++i) {
            EEPROM.write(BSSID_EEPROM_ADDRESS + i, bssid[i]);
          }
          EEPROM.commit();
        }
        
        // Hier in deiner Funktion (scan oder wie auch immer du sie nennst)
        void scanForNetwork() {
          // Ermitteln, welches Netz mit der passenden SSID das beste Signal hat
          // Das Ergebnis des Scans ist automatisch immer nach vom stärksten zum schwächsten Signal sortiert
          for (int i5 = 0; i5 < n && String(ssid) != String(WiFi.SSID(i5)); ++i5);
        
          if (i5 < n) {
            uint8_t currentBSSID[BSSID_LENGTH];
            memcpy(currentBSSID, WiFi.BSSID(i5), BSSID_LENGTH);
        
            // Speichere die BSSID im EEPROM
            saveBSSID(currentBSSID);
        
            WiFi.disconnect();
            WiFi.mode(WIFI_STA);
            initwifimanager()
            WiFi.reconnect();
          }
        }
`

what do you think about that? to complicated?

tablatronix, 2023-11-18:
Esp will save bssid for you. I did some work on this in #1342 I will revisit the status

lastphoenx, 2023-11-18:
oh - I see. thx!

lastphoenx, 2023-11-18:
> Esp will save bssid for you. I did some work on this in #1342 I will revisit the status

I tested it (your three Lines of code by putting it in first Place of My void Setup. Workshop Perfect. I Would like to Sau thank you a) for your Library and b) for your fast reply.

tablatronix, 2023-11-18:
`WiFi.setScanMethod(WIFI_ALL_CHANNEL_SCAN);`
This is the key, otherwise esp by default uses the FIRST AP not the strongest

lastphoenx, 2023-11-18:
yes, thats true. knowing this, maybe even a  ` wifiManager.setConnectTimeout(20000);` would fix it? (not best practice though)

would still be nice to know how to pass a specific BSSID to wifi.mangers autoconect or add optional BSSID paramater so that someone could fill it (like: `res=wm.autoConnect("AutoConnectAP", "password", `"BSSID")`

tablatronix, 2023-11-19:
yeah I thought I added this somewhere but I cant find it

lastphoenx, 2023-11-19:
You told that so i searched it on  here in different request, discussions but did Not find it.

lastphoenx, 2023-11-19:
It is not consistent. sometimes now it gets the best rssi but not always. it seems as if I need to make a time out.
would something like this help:   `wm.setConnectTimeout(7)` where would i place it ?

tablatronix, 2023-11-19:
I dont think so, the begin() should be doing a scan every connect before, unless you have issues like ap not found or something

lastphoenx, 2023-11-19:
thx, whatever, it would be great to pass a specifi bssid to wifimanger anyway.

tablatronix, 2023-11-19:
Yeah I will try to test this a bit, I have a mesh network, I also just updated the fastconnect feature branch to test that
Might be later this week

lastphoenx, 2023-11-19:
okay, gerat. let me know. I will help to test as i have a great mesh with 5 AP.