tzapu/WiFiManager #1679: WiFi credentials not saved after reseting/shutting down esp32.

tuadru, 2023-11-21:
## PLEASE TRY Latest Master BRANCH before submitting bugs, in case they were already fixed. ##

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32: esp32

Hardware:			esp-wroom-32/esp32 devkit v1

Core Version:      	v3.3.5-1-g85c43024c

### Description

I have tried onDemand sketch from WiFi Manager. It connects properly first time but when I reset my esp32 or turn the power off then back on it doesent connect to the network, instead I have to go into ap mode andput in credentials all over again.

### Settings in IDE

Module:  		ESP32 Dev Module	

Additional libraries: 

### Sketch

```cpp

/**
 * OnDemandConfigPortal.ino
 * example of running the configPortal AP manually, independantly from the captiveportal
 * trigger pin will start a configPortal AP for 120 seconds then turn it off.
 * 
 */
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

// select which pin will trigger the configuration portal when set to LOW
#define TRIGGER_PIN 15

int timeout = 120; // seconds to run for

void setup() {
  WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP  
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("\n Starting");
  pinMode(TRIGGER_PIN, INPUT_PULLUP);
}

void loop() {
  // is configuration portal requested?
  if ( digitalRead(TRIGGER_PIN) == LOW) {
    WiFiManager wm;    

    //reset settings - for testing
    //wm.resetSettings();
  
    // set configportal timeout
    wm.setConfigPortalTimeout(timeout);

    if (!wm.startConfigPortal("OnDemandAP")) {
      Serial.println("failed to connect and hit timeout");
      delay(3000);
      //reset and try again, or maybe put it to deep sleep
      ESP.restart();
      delay(5000);
    }

    //if you get here you have connected to the WiFi
    Serial.println("connected...yeey :)");
    

  }

  // put your main code here, to run repeatedly:
  if (WiFi.status() == WL_CONNECTED) {
  Serial.println("Works!");
  delay(1000);
  Serial.println("Works!");
  delay(1000);;
  Serial.println("Works!");
  delay(1000);
  Serial.println("Works!");
  delay(1000);
  Serial.println("Works!");
  delay(1000);
  Serial.println("Works!");
  }
}
```

### Debug Messages

```
-
```

tuadru, 2023-11-21:
P.S.
I have just used basic sketch and apparently I need auto-reconnect function to use saved credentials to reconnect.
So I must somehow combine this with button press cause I would really like to still be able to turn AP mode on demand if I want to switch wifi networks or if credential changes?

woodenplastic, 2023-11-21:
Set your    WiFiManager wm;     before Setup.

tuadru, 2023-11-22:
> Set your WiFiManager wm; before Setup.

I have put WiFi Manager wm just below int timeout = 120; 
but it still works the same, it doesen't save the credentials once I restart or power off esp.

tablatronix, 2023-11-22:
Try full flash erase, could be corrupt flash or wrong flash size?

Karthik-Official, 2024-02-09:
> > Set your WiFiManager wm; before Setup.
> 
> I have put WiFi Manager wm just below int timeout = 120; but it still works the same, it doesn't save the credentials once I restart or power off esp.

I might have a solution, but I am not sure it would work for you. Just try if you can.

Just like yours, I too had a problem with ESP not connecting to the Wi-Fi after a **reset** or **turned off** or even go inside the **config portal** and click exit.  
I have been trying many possible solutions like full flash erase my ESP, trying different Flash sizes and settings etc. 

Then I noticed:

[image]

As you can see it does remember the name of the Wi-Fi that it connected before which means in my case, it does store the Wi-Fi credentials. 

**So, I finally tried, putting ` WiFiManager wm; ` below the `int timeout = 120;` just like you,
and  `wm.autoConnect(); ` just below the `WiFi.mode(WIFI_STA);` in the setup.** 

Surprisingly, it worked, now even after restarting or power off, it just connects to the Wi-Fi credentials and got the IP as it should be.
[image]

This may or may not work; either way, trying is always better then give-up.

tablatronix, 2024-02-11:
Check your serial logs for auth failures some routers require a timeout before reconnect and you have to have a longer connecttimeout and multiple retries its very annoying

lindhardt, 2024-08-30:
You need auto connect in setup....
```
bool res = wm.autoConnect("MY_AP_NAME");
   if(!res) {
        Serial.println("Failed to connect");
        // ESP.restart();
    } 
    else {
        //if you get here you have connected to the WiFi    
        Serial.println("connected...yeey :)");
    }
```

Hosfad, 2026-11-05:
Im having the same issue, i call autoConnect in the setup function and after reboot/power loss the autoConnect function always times out.

Whenever i open the portal again it seems to be saving the ssid of the network i connected to, but its not automatically reconnecting after reboot.