tzapu/WiFiManager #1682: Not showing wifi list?

sj602, 2023-11-28:
## PLEASE TRY Latest Master BRANCH before submitting bugs, in case they were already fixed. ##

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32: 

Hardware:			ESP-12e, esp01, esp25

Core Version:      	2.4.0, staging

### Description

Is there any wifi that the wifi manager can't catch?
My mac is working well on the same wifi(=same SSID) but the wifi manager can't show the wifi to connect.

sj602, 2023-11-28:
I resolved it by changing wifi router.

closed 2023-11-28 by sj602

mindforger, 2023-11-28:
just by chance, the wifi is NOT an 5Ghz wifi but a normal 2.4Ghz ? afaik most of the 8266 can only work with 2.4Ghz

comapring a mac to an esp in this case is comparing a top tier mercedes with a smart

sj602, 2023-11-30:
It can't show both of 5Ghz and 2.4Ghz. I guess the problem was the router itself.

tablatronix, 2023-12-02:
All esp are 2.4ghz only. Almost all wifi socs are with a few expections