tzapu/WiFiManager #1688: Feature request: qrcode scanning

kamikazem, 2023-12-12:
Hi @tablatronix 
it would be nice if you could add networks by qrcode scanner , in multiple scenarios that I've witnessed the end user is not tech savvy and may not even know his wifi password , what i usually do when i'm setting up a device in their home is to ask them bring me their phone that is connected to the wifi , ask them to go to the settings and show the qrcode for that network to me ,
then i scan it with a qrcode barcode scanner , copy the password and use it to connect to the network. 
i think it would be bomb if there was a camera icon in the app that brought up a javascript barcode reader and do the rest from that.

Chalta, 2024-09-06:
Hmmm... 

To even *get* to the wifi manager portal, the user has already had to join that adhoc wifi network manually from another device. Then choosing a network from wm portal's list and typing in a password doesn't seem harder?

That said, I've used lvgl to generate dynamically a QR code on an esp32 external display. 
Scanning that code auto connects to the wifi manager adhoc portal, but that's more to save time in generating unique ssids and then commissioning multiple devices than to account for user sophistication. 

With your suggestion, it could maybe be nice if it could be maybe be two qr code scans to commission a device to a local network. But the second qr code would need to physically exist somewhere that is both secure and close to each device at time of its commissioning.

I don't know if the juice is worth the squeeze?

tablatronix, 2024-09-09:
Yeah it seems like a futile effort now that we have browser based wifi init capabilities.

kamikazem, 2024-12-27:
i tinkered out a simple barcode scanner using [Barcode Detection API](https://developer.mozilla.org/en-US/docs/Web/API/Barcode_Detection_API) , but it needs "secure context" to run.

is there any future plan for adding https support ?