tzapu/WiFiManager #1689: ESP32 fails to reset correctly after connecting to WIFI Network

serifpersia, 2023-12-19:
### Basic Infos
Latest release of WiFiManager from Arduino Library Manager

#### Hardware

Hardware: ESP32-12K(S2)

Core Version: 2.4.0

### Description
Whenever I connect from captive portal to my local network, the esp32 doesn't reset correctly, and I get issue with not being able to connect to the webserver hosted on the esp32, the browser says device refuses to connect. When I press reset button on ESP32 I can access webserver hosted on esp32 ip or mdns, does this have to spiffs I assume.
Another minor thing I don't understand is ESP32 doesn't reset when users type wrong password for their ssid, you manually have to erase wifi config, typing correct password doesn't do anything. If you could clarify how that works. 
Thanks!

### Settings in IDE

Module: NodeMCU

Additional libraries: 
#include <ESPAsyncWebServer.h>
#include <SPIFFS.h>

### Sketch
```
#include <WiFiManager.h>
#include <ESPAsyncWebServer.h>
#include <SPIFFS.h>

AsyncWebServer server(80);

void setup() {
  Serial.begin(115200);

  // Initialize SPIFFS
  if (!SPIFFS.begin(true)) {
    Serial.println("An error occurred while mounting SPIFFS");
    return;
  }

  // WiFiManager
  WiFiManager wifiManager;

  if (!wifiManager.autoConnect("AutoConnectAP")) {
    Serial.println("Failed to connect and hit timeout");
  }

  // Route to serve HTML file
  server.on("/", HTTP_GET, [](AsyncWebServerRequest * request) {
    request->send(SPIFFS, "/index.html", "text/html");
  });

  // Route to serve CSS file
  server.on("/style.css", HTTP_GET, [](AsyncWebServerRequest * request) {
    request->send(SPIFFS, "/style.css", "text/css");
  });

  // Route to serve JavaScript file
  server.on("/script.js", HTTP_GET, [](AsyncWebServerRequest * request) {
    request->send(SPIFFS, "/script.js", "application/javascript");
  });

  // Start server
  server.begin();
}

void loop() {
}
```

serifpersia, 2023-12-22:
I fixed this issue I had. I called the callback function on save config to restart esp32 board. I don't know why it doesn't do it automatically but this is the fix to be able to access the server without pressing reset button or reconnecting power to the board.

[image]

closed 2023-12-22 by serifpersia

tablatronix, 2023-12-22:
What does your logs say?
Try increasing the retries for connect, this sounds like an ap issue.

Also there is esp bug that you cannot start a new webserver check your webserver error codes, the port stays in use and cannot be released

serifpersia, 2023-12-22:
> What does your logs say? Try increasing the retries for connect, this sounds like an ap issue.
> 
> Also there is esp bug that you cannot start a new webserver check your webserver error codes, the port stays in use and cannot be released

I found the fix I'm not going to go in depth on this I just wanted to have my webserver accessible immediately after connecting to local network via wifi manager. The image above does it for me. When a password is typed and save is pressed this runs and it resets the esp32 and it I can go into the ip without issue. Thanks for making this lib its so useful. One thing I hate is not having ability to know esp32's ip after connecting. Improv web serial sdk seems like a good option to setup wifi but for now I like using your wifimanager library for wifi setup.

tablatronix, 2023-12-22:
Yeah rebooting just frees the port I think, I bet if you use a different port it works, ie 8080

serifpersia, 2023-12-22:
> Yeah rebooting just frees the port I think, I bet if you use a different port it works, ie 8080

Oh I see, well I will try different port for my webserver and see how it goes, since wifimanager webserver also uses port 80 and this bug is not releasing the port if I understood it correctly.