tzapu/WiFiManager #1693: Is it possible to retry the stored wifi SSID/PW after the Config portal has been activated?

sprior, 2023-12-31:
Sometimes the Wifi my device is configured to connect to is down temporarily, for example after a power outage my ESP8266/32 devices might boot faster than my wifi access points.  In that situation I'd like for the WifiManager config portal to come up in case the device needs to be reconfigured, but if no client connects to the config portal after a timeout the device would reboot and try the stored SSID/pw configuration again and repeat alternating between the config portal and retrying the last working value until the issue is resolved.

So my first question, is this even possible - does just starting the config portal wipe out the previous ssid/pw stored in flash?  Or is this possible and I'm just not calling things correctly (in which case I'd love an example)?

HowardsPlayPen, 2024-01-27:
The problem resonates with me - as I have an ESP32 tucked away in our attic and it has the same problem. Sometimes it just doesn't manage to reconnect to the WIFI it is configured for so it happily opens the access point and sits there until I notice it has gone rogue.

A quick scan of wifi would show that the stored details are for an available wifi so as mentioned above, periodically retrying this wifi rather than defaulting to access point would be amazing.

I am changing my code to use PsychicHttp for OTA updates and storing the wifi credentials in a json file that I bundle into the updated firmware - and removing use of this library for now.

tablatronix, 2024-01-27:
Make sure you have a configportaltimeout set and bump your connectreries and conecttimeouts

closed 2024-01-27 by tablatronix

sprior, 2024-01-27:
@tablatronix Since I'm the one who asked the question I'm not sure what the answer was.  So are you saying that if I reboot after portaltimeout that it'll retry again with the previously saved wifi credentials or at that point are those credentials gone and someone will have to connect through the portal and reconfigure the wifi?

tablatronix, 2024-01-28:
credentials are never reset automatically

sprior, 2024-01-28:
So you're saying that once WifiManager goes into portal mode it's still possible to reboot after timeout and attempt to reconnect using the previously working credentials?

tablatronix, 2024-01-28:
Yeah of course