tzapu/WiFiManager #1696: Wifi credentials

Guytou11, 2024-01-03:
After a few months working without any issue, WiFiManager lost the WiFi credentials. They have been put in again through the portal and the ESP32 is working fine again.  Any reason why WiFiManager lost these credentials ?

Chalta, 2024-09-10:
_Did_ wifi manager lose the credentials? How are you sure this is what happened? Perhaps there was a network change? Or the ESP32 flash was erased?

Guytou11, 2024-09-10:
Hi, There was no network change.  Not sure how the ESP32 flash could have been erased. The ESP32 did still all what it was intented to do, exept that he didn't remember the WiFi credentials. It happened only once, but I am still interested to learn why.

tablatronix, 2024-09-10:
Cosmic rays

loloz3, 2024-10-06:
My ESP8266 is also losing Wifi credentials when there is short power disturbance...
(it is install on a Archery Bow and there are a lot of vibrations)
memcpy is currently used to save them, is it possible to changed and used LitteFS ?

B1ackt34, 2024-12-15:
> My ESP8266 is also losing Wifi credentials when there is short power disturbance... (it is install on a Archery Bow and there are a lot of vibrations) memcpy is currently used to save them, is it possible to changed and used LitteFS ?

I'm so curious about your installation ! What does an ESP on a bow?

tablatronix, 2024-12-15:
Wifi creds are stored in flash by the esp lib, if you are losing them something is wrong with your memory or board

loloz3, 2024-12-15:
I tested with several ESP8266… and all have the same issue. Currently I must use wifi credentials in my code… I cannot use WifiManager any more ( this is happened too often )
FYI Here my project (in French sorry)
https://loloz3.ddns.net/ArcherTrainer/build.html#hard