tzapu/WiFiManager #1703: ESP32-S3 WiFi credentials lost after reset

mikebaylis, 2024-01-16:
**ESP32-S3 WiFi credentials lost after reset.**

Using https://github.com/tzapu/WiFiManager.git

Code:
wifiManager.autoConnect("myAP", "myPassword");

WiFiManager.autoConnect() always goes to AP captive portal after reset on ESP32-S3.
Why is this?  I see call to esp_wifi_get_config(WIFI_IF_STA, &conf) returns an empty SSID string.  The return status code for the call to esp_wifi_get_config(WIFI_IF_STA, &conf) is not checked, but if I add esp_err_t result = esp_wifi_get_config(WIFI_IF_STA, &conf) I see result = ESP_ERR_WIFI_NOT_INIT
And why is this?  I find 
#elif defined(ESP32)
	// wifiLowLevelInit(persistent); // @todo fix for S3, not found
So I uncomment this line, and sure enough the call to wifiLowLevelInit() is not found - wifiLowLevelInit() is in WiFiGeneric.cpp but not made public.

And I find this issue reported to team Arduino:
https://github.com/espressif/arduino-esp32/issues/9056
which ends with comment "I'll give you the nvs keys next year." Is this a joke?  I don't know.

Anyway, there is an easy fix which is to add a call to WiFi.mode(WIFI_AP_STA) before calling WiFiManager.autoConnect().

This issue does not occur on ESP32, only on ESP32-S3, so it seems WiFi driver is enabled by default at startup for ESP32, but not so for ESP32-S3.

tablatronix, 2024-01-17:
Yeah this is a wifi init issue I may have accidentally renabled this bug let me check

sapzxc, 2024-02-11:
Thanks a lot!
I used this library first time and attempted to save credentials manually until I came across your ticket. I had assumed there was no save functionality. Upon further investigation of the code, I couldn't find any save functions. Even in the example functions, there were no save functions. In the readme, I only found references to playing with flags.

It would be beneficial to include some additional text in the README for new users, such as "Credentials are automatically saved to independent memory and restored after power loss. If your settings resets, check issue #1703."