tzapu/WiFiManager #1709: 192.168.4.1 : blank page

davidmarli, 2024-02-10:
I installed wifimanager on an esp32.
All went fine, I configured it, it can connect to internet.
10 days later, the device rebooted, but can't connect to my wifi.
It starts de webportal procedure.
So I connect to the AP. But when I write 192.168.4.1, I just have a blank page.
I try 192.168.4.1/restart : doesn't do anything
192.168.4.1/info : nothing

What's going wrong ?

Thank you

tablatronix, 2024-02-11:
No idea sounds like a fatal failure or corrupt memory. You would have to connect serial to see

davidmarli, 2024-02-11:
Ok, it's something I sell. The buyer will sent it back to me.
I will check what is wrong.

tablatronix, 2024-02-12:
Yeah never saw this before, not a common thing