tzapu/WiFiManager #1711: WiFi settings are not saved if ESP32 is restarted in SaveConfigCallback

rusty-labs, 2024-02-14:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32: 

Hardware:			Heltec wifi-lora-32-v3

Core Version:      	2.4.0, staging

### Description

Steps to reproduce

- Connect to AP
- Set incorrect wifi password and save
- Add setSaveConfigCallback to code
- Restart ESP32 
- Connect to AP
- Provide correct password and save
- Web shows "Saved" and ESP32 is restarted
- ESP32 is trying to connect to previously provided, incorrect password

### Settings in IDE

Module:  			Heltec wifi-lora-32-v3

### Sketch

```cpp

void saveConfigCallback()
    {
        delay(2000);
        ESP.restart();
    }

        WiFi.mode(WIFI_STA);

        wm.setSaveConfigCallback([&]()
                                 { saveConfigCallback(); });

        wm.setBreakAfterConfig(true);

        wm.addParameter(&paramMqttServer);
        wm.addParameter(&paramMqttPort);
        wm.addParameter(&paramMqttUser);
        wm.addParameter(&paramMqttPassword);

        std::vector<const char *> wm_menu = {"wifi", "exit"};
        wm.setShowInfoUpdate(false);
        wm.setShowInfoErase(false);
        wm.setMenu(wm_menu);

        bool res = wm.autoConnect(_SSID);
```

tablatronix, 2024-02-14:
I suspect breakafterconfig i will test

rusty-labs, 2024-02-14:
Yeah, if I don't use setBreakAfterConfig my callback is not even called

rusty-labs, 2024-02-15:
Ok there're multiple issues

Settings are not saved if no successful connection happened,
```
  WiFi.persistent(true);
  ret = WiFi.begin(ssid.c_str(), pass.c_str(), 0, NULL, connect);
  WiFi.persistent(false);
```
So I toggled off `wm.setSaveConnect(false);`
I assume if user saves settings, settings should be saved no matter what, and connection procedure should be out of connection block

```
WiFi.persistent(true);
ret = WiFi.begin(ssid.c_str(), pass.c_str(), 0, NULL, false);
WiFi.persistent(false);
// Try to connect here
```

Other issue is within the UI
Opening configuration portal second time shows my  previously selected SSID in gray color and I assumed I don't need to select it again, so I changed only password.
However on the next bootup SSID is empty while the updated password is there

rusty-labs, 2024-02-15:
I uploaded the fix addressing UI issue and saving empty ssid

tablatronix, 2024-02-15:
Yeah there is a other issue about that thanks
Its already saves if its placeholder. I will think of a better way to represent this like disable until clicked or something

rusty-labs, 2024-02-15:
For the sake of consistency it'd be better to use value instead of a placeholder, since custom parameters are using values
I also checked behavior on my router in client mode, and it also keeps SSID as a value even if it's wrong

tablatronix, 2024-02-15:
Custom params can also use placeholders

tablatronix, 2024-02-15:
I used placeholder because I do not want wifi args resubmitting if only changing params. This way they do not get sent in plain text everytime you save, if not needed. Also we cannot populate password.

hmm I thought this fixed this

9df7acdeeb6a7f614cdcba4a3f8c117ab7c23718

rusty-labs, 2024-02-15:
Sending SSID in a plain text is totally fine.
#
> saveConfigCallback() and not using return or status? Just Curious
Yeah, I can restart after connection complete too, I just prefer callbacks, in case if I want to use non-blocking mode etc.

I believe the root cause of these problems is that WiFi_SSID and _ssid are not synced, should be setter/getter instead

rusty-labs, 2024-02-16:
Password has the same problem

- Provide correct ssid and pass, save
- Load configuration portal again
- Change custom parameters
- Save
- Can't connect to WiFi because password is gone

rusty-labs, 2024-02-16:
Ok I fixed all the issues above in my fork

https://github.com/rusty-labs/WiFiManager/blob/981a540793fd1c1b501be7f6071da42fe5aaa3de/WiFiManager.cpp#L2110C3-L2116C4

https://github.com/rusty-labs/WiFiManager/blob/981a540793fd1c1b501be7f6071da42fe5aaa3de/wm_strings_en.h#L66

https://github.com/rusty-labs/WiFiManager/blob/981a540793fd1c1b501be7f6071da42fe5aaa3de/WiFiManager.cpp#L1293C3-L1303C4

I did some other fixes but I think these 3 should be sufficient to solve the problem

tablatronix, 2024-02-16:
What browser? Ssid and pass empty should do nothing

rusty-labs, 2024-02-17:
It has nothing to do with a browser

1. _ssid = server->arg(F("s")); returns empty ssid, because it's a placeholder not a value
2. handleWifiSave() makes _ssid empty
3. connect=true
4. processConfigPortal saves empty _ssid to flash

tablatronix, 2024-02-17:
I still dont understand this issue, or see it..

```
      // skip wifi if no ssid
      if(_ssid == ""){
        #ifdef WM_DEBUG_LEVEL
        DEBUG_WM(WM_DEBUG_VERBOSE,F("No ssid, skipping wifi save"));
        #endif
      }
```

rusty-labs, 2024-02-17:
Ok I got your latest changes
Your fix doesn't work if I provide incorrect SSID and correct Password first, and then after restart provide correct SSID
But for the steps I described above it works

Btw this code shouldn't be called if _connectonsave = false
https://github.com/tzapu/WiFiManager/blob/d8f8d5478a646a9ec14d0418aa0c4b96c1899b58/WiFiManager.cpp#L1038C3-L1043C4
Also I don't understand why 
`if(_aggresiveReconn && _connectRetries<4) _connectRetries=4;`
If I set _connectRetries = 2, why all of a sudden it becomes 4 ? Why do you even need _aggresiveReconn  ?

Overall the logic is becoming too tangled, the code needs a cleanup, too many states contradicting each other

tablatronix, 2024-02-17:
It says it in the source

```
    bool          _aggresiveReconn        = true; // use an agrressive reconnect strategy, WILL delay conxs
                                                   // on some conn failure modes will add delays and many retries to work around esp and ap bugs, ie, anti de-auth protections
                                                   // https://github.com/tzapu/WiFiManager/issues/1067
```

tablatronix, 2024-02-17:
I will remove it and see if the bug is fixed, and see if I can move this into a workaround check for specific versions and or failure codes.

rusty-labs, 2024-02-19:
_aggresiveReconn is always true and there is no public setter for this, instead you can just 
`_connectRetries = max(connectRetries, 4);` at least the logic would be explicitly expressed instead of  some hidden bool state 
But I'd get rid of it too, those who have troubles with a connection can increase number of retries in a client code
The library code shouldn't fix specific problems instead provide an interface to do so
Also delay(1000); should be after wifiConnectNew, otherwise execution is just delayed for nothing

tablatronix, 2024-02-19:
The delay is necessary for this failure mode

rusty-labs, 2024-02-19:
Delay yes, but _aggresiveReconn no, if it's always true

rusty-labs, 2024-02-19:
Btw delay not needed, you already have _connectTimeout which is not used, set _connectTimeout = 1000 and it will work as a delay

tablatronix, 2024-02-20:
They function differently. Timeout is not the same as delay for reconnect. 
I added both those options that test was implemented as a result of its feedback.

i just need to decide how to reimplement it better so its not always required.

real-bombinho, 2024-03-09:
I have version 2.0.17 and it seems this is not fixed in this version by the looks of? 

I am just porting some working stuff from ESP8266 to the ESP32 and it keeps forgetting the WiFi credentials on each restart.

tablatronix, 2024-03-10:
create a new issue with information, unrelated to this issue

tablatronix, 2024-03-10:
> Your fix doesn't work if I provide incorrect SSID and correct Password first, and then after restart provide correct SSID
But for the steps I described above it works

Is this really a issue? who does this?

tablatronix, 2024-03-10:
Closing unless you can provide how to reproduce and provide logs. I tested this as much as I can and have no issue

closed 2024-03-10 by tablatronix