tzapu/WiFiManager #1713: Add support for simultaneo use whit ESPAsyncWebServer.h

AnthonyAC23, 2024-02-17:
when you want to use ESPAsyncWebServer.h simultaneously with this library since using both at the same time generates errors due to redeclarations, etc.

please add this extension. thanks

some debug messages.
/http_parser.h:95:6: error: redeclaration of 'HTTP_DELETE'
XX(0,  DELETE,      DELETE)       \
/http_parser.h:138:45: note: in definition of macro 'XX'
#define XX(num, name, string) HTTP_##name = num,
/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
HTTP_METHOD_MAP(XX)
ESPAsyncWebServer.h:66:3: note: previous declaration 'WebRequestMethod HTTP_DELETE'
HTTP_DELETE  = 0b00000100,
/http_parser.h:96:6: error: redeclaration of 'HTTP_GET'
XX(1,  GET,         GET)          \
/http_parser.h:138:45: note: in definition of macro 'XX'
#define XX(num, name, string) HTTP_##name = num,
/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
HTTP_METHOD_MAP(XX)
/ESPAsyncWebServer.h:64:3: note: previous declaration 'WebRequestMethod HTTP_GET'
HTTP_GET     = 0b00000001,
/http_parser.h:97:6: error: redeclaration of 'HTTP_HEAD'
XX(2,  HEAD,        HEAD)         \
/http_parser.h:138:45: note: in definition of macro 'XX'
#define XX(num, name, string) HTTP_##name = num,
/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
HTTP_METHOD_MAP(XX)

etc......

tablatronix, 2024-02-17:
How to avoid this in dependancy libs? Also there is a esp port issue with running 2 services

lucaoliano, 2024-03-09:
As workaround in the file where you include ESPAsyncwebServer.h you must include WebServer.h first like below

```
#include <WebServer.h>
#include <ESPAsyncwebServer.h>
```

AnthonyAC23, 2024-03-09:
Tanks @lucaoliano, just like you said, that solved the problem.

AnthonyAC23, 2024-03-09:
Works!

closed 2024-03-09 by AnthonyAC23