tzapu/WiFiManager #1717: Avoid compiler warning. Wrong specifier used for printf statement.

HenkHoldijk, 2024-03-03:
Compiling for the ESP32-C3 using the Arduino IDE 1.8.19 results in a compiler warning for WiFiManager.cpp on line 3368.

Original line : "_debugPort.printf("[MEM] free: %5d | max: %5d | frag: %3d%% \n", free, max, frag);"

The %5d does not match the uint32_t free type.

Replace with : "_debugPort.printf("[MEM] free: %5ld | max: %5d | frag: %3d%% \n", free, max, frag);"

tablatronix, 2024-03-03:
Hmm I would think free and max should be the same type also

tablatronix, 2024-03-03:
https://github.com/espressif/arduino-esp32/blob/aed7b4fa598765db61678d3253401b8cd64a8ec8/cores/esp32/chip-debug-report.cpp#L25

HenkHoldijk, 2024-03-03:
Agree regarding the max. The frag is OK.

Currently they are declared as:
    uint32_t free = info.total_free_bytes;
    uint16_t max  = info.largest_free_block;
    uint8_t frag = 100 - (max * 100) / free;

Should be:
    uint32_t free = info.total_free_bytes;
    uint32_t max  = info.largest_free_block;
    uint8_t frag = 100 - (max * 100) / free;

The printf then becomes : _debugPort.printf("[MEM] free: %5ld | max: %5ld | frag: %3d%% \n", free, max, frag);